AlbrechtL/welle.io #97: Options

Chris805, 2017-07-01:
hallo Albrecht,

I downloaded Welle.

Very nice program. i want to use it beacause of the zadig driver

but can i ask a question?

I miss the ensemblenames and channels  in the player... i have to search the channel list.
Also a protection level info would be great.

is this something you are going to implement or not?

Also a signal walking graph would be great.

greetings, Chris

AlbrechtL, 2017-07-01:
If I understood you correct, you would like to see the following status information inside the expert view:
* Ensemble name
* Channel name
* Protection level
* Walking graph

Can you explain the use cases more in detail why you need these information displayed?

Chris805, 2017-07-02:
hello,

beacause it makes it a lot more user friendly.

I think more people would use it then.

In all other dab software it is also displayes.

Also it would be nice to attach a volume bar en sort the channels on ensemble.

Greetz

AlbrechtL, 2017-07-25:
I added your request to the [open task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

closed 2017-07-25 by AlbrechtL