AlbrechtL/welle.io #98: No display shown after introduction of TouchComboBox

russel, 2017-07-02:
The changeset at f8f2f7c69ec84d64a2dc3cf99f9c3a526e573843 introduces a change in the QML code such that whilst the CMake build completes without error on execution:
```
QQmlApplicationEngine failed to load component
qrc:/src/gui/QML/main.qml:111 Type SettingsPage unavailable
qrc:/src/gui/QML/SettingsPage.qml:206 Type TouchComboBox unavailable
qrc:/src/gui/QML/style/TouchComboBox.qml:2 module "QtQuick.Controls" version 2.2 is not installed
```
This is on Debian Sid with both related packages (qml-module-qtquick-controls and qml-module-qtquick-controls2) installed. I am assuming this is a problem because Debian Sid is still using Qt 5.7.1.

AlbrechtL, 2017-07-02:
Fixed in 92dac1d.

closed 2017-07-02 by AlbrechtL

russel, 2017-07-02:
Works for me now, thanks.