AlbrechtL/welle.io #100: Win10: Uninstall only visible for admin

andimik, 2017-07-03:
This seems to be the first software which does not appear in the apps menu for a user.

[image]

--------------------

Only the admin has an entry.

[image]

AlbrechtL, 2018-01-26:
I don't known the reason for that. It would be nice if someone can check how to configure the [Qt Installer Framework](https://doc.qt.io/qtinstallerframework/)  to solve this issue.

AlbrechtL, 2018-09-13:
@andimik Can you try the latest Windows installer please? Because I switched from Qt Installer Framework (QTIFW) 2.0 to QTIFW 3.0.

andimik, 2018-09-20:
I cannot check in the next days. Let me try it in Windows in October again with multiple users. Currently I am working in Ubuntu only.

andimik, 2018-12-31:
No, same as in July. I thought you have changed something, but the uninstall is still only visible for the admin.

I've installed the latest nightly build (2 days old), 20181229_29f4d4a_Windows_welle-io-setup.exe

for the admin
[image]

for the user
[image]

AlbrechtL, 2019-01-07:
Did you install it as admin or user?

andimik, 2019-01-07:
As user which then asked for the admin password.

AlbrechtL, 2022-12-16:
Can you test it again with the latest nightly build (https://welle-io-nightlies.albrechtloh.de/). The background is because I changed the installer from Qt Install Framework to InnoSetup.

AlbrechtL, 2024-10-08:
I'm closing this issue because I'm not sure if it is still relevant. If it still occurs in the latest version, we can reopen it.

closed 2024-10-08 by AlbrechtL