AlbrechtL/welle.io #101: Compilation error

zygmund2000, 2017-07-06:
../welle.io/src/input/CAirspy.cpp: In constructor ‘CAirspy::CAirspy()’:
../welle.io/src/input/CAirspy.cpp:78:38: error: call of overloaded ‘abs(uint32_t)’ is ambiguous
         if (abs(myBuffer[i] - 2048000) < distance) {
                                      ^
In file included from /usr/include/c++/7.1.1/cstdlib:75:0,
                 from /usr/include/c++/7.1.1/bits/stl_algo.h:59,
                 from /usr/include/c++/7.1.1/algorithm:62,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qglobal.h:109,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qalgorithms.h:43,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qdebug.h:44,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/QDebug:1,
                 from ../welle.io/src/input/CAirspy.cpp:32:
/usr/include/stdlib.h:751:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/7.1.1/cstdlib:77:0,
                 from /usr/include/c++/7.1.1/bits/stl_algo.h:59,
                 from /usr/include/c++/7.1.1/algorithm:62,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qglobal.h:109,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qalgorithms.h:43,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qdebug.h:44,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/QDebug:1,
                 from ../welle.io/src/input/CAirspy.cpp:32:
/usr/include/c++/7.1.1/bits/std_abs.h:56:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:61:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:70:3: note: candidate: constexpr double std::abs(double)
   abs(double __x)
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:74:3: note: candidate: constexpr float std::abs(float)
   abs(float __x)
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:78:3: note: candidate: constexpr long double std::abs(long double)
   abs(long double __x)
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:84:3: note: candidate: constexpr __int128 std::abs(__int128)
   abs(__GLIBCXX_TYPE_INT_N_0 __x) { return __x >= 0 ? __x : -__x; }
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:102:3: note: candidate: constexpr __float128 std::abs(__float128)
   abs(__float128 __x)
   ^~~
../welle.io/src/input/CAirspy.cpp:79:49: error: call of overloaded ‘abs(uint32_t)’ is ambiguous
             distance = abs(myBuffer[i] - 2048000);
                                                 ^
In file included from /usr/include/c++/7.1.1/cstdlib:75:0,
                 from /usr/include/c++/7.1.1/bits/stl_algo.h:59,
                 from /usr/include/c++/7.1.1/algorithm:62,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qglobal.h:109,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qalgorithms.h:43,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qdebug.h:44,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/QDebug:1,
                 from ../welle.io/src/input/CAirspy.cpp:32:
/usr/include/stdlib.h:751:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/7.1.1/cstdlib:77:0,
                 from /usr/include/c++/7.1.1/bits/stl_algo.h:59,
                 from /usr/include/c++/7.1.1/algorithm:62,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qglobal.h:109,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qalgorithms.h:43,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/qdebug.h:44,
                 from ../Qt/5.9.1/gcc_64/include/QtCore/QDebug:1,
                 from ../welle.io/src/input/CAirspy.cpp:32:
/usr/include/c++/7.1.1/bits/std_abs.h:56:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:61:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:70:3: note: candidate: constexpr double std::abs(double)
   abs(double __x)
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:74:3: note: candidate: constexpr float std::abs(float)
   abs(float __x)
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:78:3: note: candidate: constexpr long double std::abs(long double)
   abs(long double __x)
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:84:3: note: candidate: constexpr __int128 std::abs(__int128)
   abs(__GLIBCXX_TYPE_INT_N_0 __x) { return __x >= 0 ? __x : -__x; }
   ^~~
/usr/include/c++/7.1.1/bits/std_abs.h:102:3: note: candidate: constexpr __float128 std::abs(__float128)
   abs(__float128 __x)
   ^~~
make: *** [Makefile:7902: CAirspy.o] Error 1

AlbrechtL, 2017-07-08:
Which Linux and build mechanism (QTcreator or cmake) did you use?

zygmund2000, 2017-07-11:
I have fresh install Arch/Manjaro 64 with qt5.9.1 downloaded from link: https://www.qt.io/download-open-source
All packages are in latests versions, mechanism QTcreator.

zygmund2000, 2017-07-11:
This problem is only with airspy library, wihout is ok.

AlbrechtL, 2017-07-11:
Please try to replace `abs(...)` by `std::abs(...)` in [CAirspy.cpp line 78 and 79](https://github.com/AlbrechtL/welle.io/blob/d527eb40d9ec549dded88dcba388596bf16340cc/src/input/CAirspy.cpp#L78) and report your result.

AlbrechtL, 2017-07-25:
@zygmund2000 Did you have any results?

AlbrechtL, 2017-08-07:
I'm closing this issue because there are not more information and I can't reproduce it.

closed 2017-08-07 by AlbrechtL

zygmund2000, 2017-08-11:
Arch based systems like Manjaro have newer version of all packages than other distros like Ubuntu, so I you don't have it you cant reproduce issue.

AlbrechtL, 2017-08-12:
Thanks to Matthias Braendli this issue is fixed in version 35b529b.