AlbrechtL/welle.io #102: [Discussion] Replacing exisiting DAB/DAB+ code with sdrdab code

AlbrechtL, 2017-07-08:
The [AGH](http://www.agh.edu.pl/en/) University of Science and Technology in Kraków developed a DAB decoder [sdrdab](https://sdr.kt.agh.edu.pl/sdrdab-decoder/). It seems that this the code us much more readable with better algorithms (SNR, AGC, BER etc.) than the existing welle.io code.

AlbrechtL, 2017-07-09:
**To everyone**: Please compare [sdrdab](https://sdr.kt.agh.edu.pl/sdrdab-decoder/) with welle.io by means of decoder performance. Which parts are worth to use?   

Write your results in here.

viktorgino, 2017-07-10:
I think one of us should get in touch with the developers of that project (if haven't already), because from what I can tell they want a better UI, whereas we want a better back-end, so might worth combining efforts.

I think it would be beneficial to some users that we keep support for AirSpy and SoapySDR. From a quick glance through the code we have the AbstractDataFeeder class, so adding an other tuner would be as difficult as extending it (I'd assume that the way get a list of tuner and change them, both in GUI and CLI need to be changed slightly).

Also does this work on ARM (I see mention of an arm-developement branch, but could find it on their site)? I'll try to test it on the Pi 3, but would be good to hear others experience before.

AlbrechtL, 2017-07-10:
I already contacted the developers and Jarosław Bułat (kwant@agh.edu.pl) responded.
He wrote me that sdrdab is designed as a library and we can use it. 
> The aim of our project (sdrdab) is to develop DAB+ decoder as a library. It
was designed to integrate it into other projects, like welle.io, VLC, etc...
 From this point of view it perfectly matches your idea.

They are also interested in somebody that uses sdrdab 
> Any feedback and suggestions are welcome

I ask them about the performance, synchronization and OS support. Here is the full answer:
> Time synchronization is quite good: we are making use of precise two consecutive NULL symbol detection: https://sdr.kt.agh.edu.pl/sdrdab-decoder/null.html calculation real length of transmission frame and performing resampling to adjust it. We use hardware PLL adjusting in case of using RTL dongle or libresampling if we read data from file (the second option is quite inefficient and should be rewritten).
>
> Frequency shifting ("fc" in description) is detected very precisely on the basis of Phase Reference symbol: https://sdr.kt.agh.edu.pl/sdrdab-decoder/phase_reference.html However, convergence of correction is slow. In case of big initial error (frequency shift) it converge slow. Correction is done by remodulation: e^(-j*2*pi*...), we initially use direct Euler equation (sin(...)+jcos(...)) however it was very slow, then we test and successfully apply LUT technique for sin() and cos() calculation. And now it is fast We still working of better logic for faster convergence of this algorithm.
>
> [...] It is hard to say how fast it is It was tested on Intel Core i7-3520M, 2.9 GHz (2 core, 4 threads). It switch between p-state 3.4 GHz (Turbo) and 1.5 GHz. The most CPU extensive task are preformed in DataDecoder, it cover task between (including) time deinterleaver and super frame (AAC) processing. AAC decoding is done in other thread, demodulation and frequency deinterleaver is done in other thread as well. From htop you see, DataDecoder thread consume 71% of single core. Much of this is resources is consumed by convolutional decoder. We use custom Viterbi code, optimized with SSE3 (NEON on ARM), and it is bottleneck of whole decoder. Viterbi decoder is optimal but slow :-/ Maybe in October one of student finish Viterbi-spiral implementation. It should be faster. BCJR shoud be the best choose for this problem but nobody want to do that :-/ Finaly, rewriting this algorithm should reduce computational complexity 2-3 times.
>
> In other part of code we try to minimize resources. For example we do not decode OFDM symbols that are not necessary for playing given station. Only symbols that carry data for given station are demodulated and processed. Memory footprint is reduced as much as possible. For example we optimize data processing to reuse the same buffers instead of copy it between consecutive stage of algorithms. This, unfortunately, decrease readability of the code
> 
> [...]
> In the README you find how to build and Run. We create "sdrtool" script as a facility for build, debug, test, download data, etc...Read instruction, it is helpful . We do not have installer. Library is build to build/src/sdrdab/libsdrdab.a. It is example of use: sdrdab-cli. It is compiling by default to build/bin/sdrdab-cli. Do not use sdrdab-gui, it is crappy :-/ Sdrdab was develop and tested on Ubuntu 16.04 or newer (older one need upgrading CMake). Some student port version 1 on RaspberryPI v2. It was (hardly) working on overclocked RPi So... it is possible, however, this port was never merge to master branch so it is not available for version 2 and 3 (most critical part was optimizing Viterbi with help of NEON instruction - I have this code so in future we try to recover ARM port). Mac OS port was done for version 2 and is integrated with code. It was compiling on Mac but was not tested. No one wants to port it to Windows :-0, however, it should be possible.

He also sent me the sdrdab 3.0 beta which is not public yet.

viktorgino, 2017-07-10:
How different is the new version from the one that's currently public?

I think it would be interesting to see if we could take parts of their code that are more efficient than what we have now.

zygmund2000, 2017-07-11:
I have tried sdrdab 2.0 but it can't receive DAB signal from rtl2838, some prints in console but nothing else, no synchronisation.

andimik, 2017-07-11:
I suppose you have a good reception with other programs, haven't you?

zygmund2000, 2017-07-11:
Yes I have, eg. qt-dab works ok.

viktorgino, 2017-07-11:
@zygmund2000 Can you upload you console output as a file, so we can see what's happening

zygmund2000, 2017-07-11:
./sdrdab-gui 
qt5ct: using qt5ct plugin
QObject::startTimer: Timers cannot be started from another thread
Found Rafael Micro R820T tuner
WARNING: Failed to reset buffers.
first_null: 28542, second_null: 55599, frame_size: 27057
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
first_null: 22733, second_null: 45146, frame_size: 22413
first_null: 31620, second_null: 56000, frame_size: 24380
first_null: 21918, second_null: 52636, frame_size: 30718
first_null: 25329, second_null: 52372, frame_size: 27043
......

sdrdab-cli crashes after start

viktorgino, 2017-07-11:
Interesting. I'll try with the same tuner tonight.

zygmund2000, 2017-07-11:
long -> null: 10252, min_energy: 38,438782, max_energy: 45,462723

******************* NULL not detected *************************

AlbrechtL, 2017-08-09:
The sdrdab code is available here https://github.com/kwanty/sdrdab now. Jarosław also wrote me that version 3 will be available soon.

twoofseven, 2017-09-28:
Any news regarding version 3?

andimik, 2017-09-29:
I guess you have to ask the Polish guys.

AlbrechtL, 2017-10-03:
I asked Jarosław via e-mail a few days ago but no reply yet. Anyway I started integrating version 2.

AlbrechtL, 2017-11-16:
Version 3 beta is now online at the branch ["v3beta"](https://github.com/kwanty/sdrdab/tree/v3beta).

AlbrechtL, 2017-11-23:
A prototype integration of sdrdab can be found inside the branch ["next"](https://github.com/AlbrechtL/welle.io/tree/next). This version is roughly working under Linux but with a lot of issues. Please see https://github.com/kwanty/sdrdab/issues/3 for details.

Beside the sdrdab issues there are some open points in [TODO](https://github.com/AlbrechtL/welle.io/blob/next/TODO).

AlbrechtL, 2018-01-26:
I close this issue for now because we don't have plans to switch to sdrdab soon.

closed 2018-01-26 by AlbrechtL

AlbrechtL, 2018-02-13:
I renamed the branch to [sdrdab-test](https://github.com/AlbrechtL/welle.io/tree/sdrdab-test).