AlbrechtL/welle.io #103: [Android] Android Auto App Integration

twoofseven, 2017-07-11:
As described here [Building Apps for Auto](https://developer.android.com/training/auto/index.html ) and [Providing Audio Playback for Auto](https://developer.android.com/training/auto/audio/index.html)
media applications can integrate within Android Auto.

I personally use the Android Auto as standalone app running on my Android Head-Unit and like to be able to use welle.io's DAB functionality while Android Auto is running in Car mode. Furthermore the Android Auto UI is specifically designed to be used while driving (e.g. font sizes, colour contrast, restricted feature set, ...)

I already started to implement a basic support [here](https://github.com/twoofseven/welle.io/commit/bf85720ba58e199e7bf0ffc5bbef758f218e078b)

Upon the user plugs an RTL-SDR device, welle.io starts as a background service and shows a notification with basic media control functionality to play/pause, skip to next station and start a channel scan.

[image]

Within the Android Auto app the common play/pause, skip next/previous station support is available:

[image]

In addition the channel scan (working) and switch to next channel (not yet working) is available as custom action:

[image]

And DAB stations can be selected from a stations list:

[image]

### Issues

To enable welle.io app to run in background, it was necessary to integrate the RTL-SDR & libusb library, because I am not aware of a solution to send an intent from a Qt service to an Android activity. I have workarounds in mind to use the Android RTL-SDR driver app again, but have not had time to implement them yet. => Further feedback is highly welcome.

In addition, the current implementation does not handle “Pause” requests as expected and shows the error message “RTL-SDR unplugged”:

[image]

This is cause by calling the Stop() method of the RTL_SDR Device to reduce the CPU load while the app is not playing.

### Other features to be implemented

1. Separation of the main activity and background service as described in [Qt on Android: How to create an Android service using Qt](https://www.kdab.com/qt-android-create-android-service-using-qt/)
2. Support for background images (MOTImage) within Android Auto app. 
3. Channel listing within Android Auto to directly select an DAB ensemble and the support to scan for the next available DAB ensemble rather than the complete channel scan (as described above). 
4. Favourite stations list within Android Auto, where favourite stations will not be deleted by a channel scan. A favourite stations list within the Qt context and a custom action for Android Auto is already implemented, but I have not implemented the respective JNI calls yet. 
5. Recording of DAB streams, controlled via a custom action within Android Auto. The custom action is already prepared, but stream recording is in general not yet supported within welle.io. 
6. Better icons within Android Auto and the Android notification. Currently Android's built-in icons are used.

AlbrechtL, 2017-09-05:
A first test version can be downloaded [here](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/4308fa90d5827254c6932f8b61f4bd7e6e591b95/20170905_Android_armv7_welle-io.apk).

@twoofseven Please provide a small instruction how to use it

twoofseven, 2017-09-06:
Here the changes in detail:

### General

- Add ensemble name in Radio view and rearrange Radio view items to be able to render the additional ensemble name
- Add Hardware AGC switch in Settings view (expert mode only).
- Add automatic playback of first found station for manual channel switching. In case respective stations are already available for the selected channel, the first station will be played automatically.
- Always show manual channel switching. In addition the combobox will always show the current channel.
- Add percentage information during channel scan
- Remove GUI update timer and use the GUI and RadioController properties within QML directly. => Lower latency
- Centralise Station List handling within RadioController and extended StationList class (save & load stations)
- Replace QPixmap by QImage within RadioController. Otherwise the Android background service always crashes due to allocation of QPixmap
- Handling of spectrum / FFT calculation changed due to separation of Android activity (GUI) and Android background service (RadioController)

### Android

- Support of latest Android RTL-SDR driver changes to start automatically upon the RTL-SDR device got connected [https://github.com/martinmarinov/rtl_tcp_andro-/pull/15](url)
- Background Audio playback including IPC via QtRemoteOjects between activity and service as requested in #88 
- Media Integration to support the following DAB functionality via notification, lock screen, Bluetooth*, Android Auto and most likely Android Watch (not tested, due to missing device):

1. DAB metadata (dab status, current station and radio text)
2. DAB MOT SlideShow
3. DAB command and control (play/pause, skip next/previous station, start/stop channel scan)
4. DAB channel and station list
5. Favourite stations

* Bluetooth support varies depending on feature set of connected Head-Unit and Android version

twoofseven, 2017-09-06:
Here some screenshots of the latest version:

[image]

[image]

[image]

[image]

[image]

AlbrechtL, 2017-09-17:
You can try a new test version [20170917_Android_armv7_welle-io.apk](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/44974c20fcf182e9552fb021a6a98326c9403810/20170917_Android_armv7_welle-io.apk).

AlbrechtL, 2017-11-24:
Android Auto will be part of the next official release. All changes are already merged into the master branch.

closed 2017-11-24 by AlbrechtL

twoofseven, 2017-11-24:
Cool.

Matthias

AlbrechtL, 2018-07-17:
I assume you are using a smart phone with only one connector. The rtl-sdr dongle has to be connected to the USB port and then you can't use the HDMI interface.