AlbrechtL/welle.io #104: Font too large

andimik, 2017-07-12:
I cannot read the channel number as the font under Android is too large.

[image]

andimik, 2017-07-14:
Or alternatively check, why the border is so thick.

andimik, 2017-08-13:
Try (but not tested)
                        bottomPadding: x
                        topPadding: x
                        padding: x

or 

                        Layout.preferredHeight: x
                        Layout.preferredWidth: x
                        Layout.minimumHeight: x
                        Layout.minimumWidth: x

And for the touch buttons maybe add

                        antialiasing: true

as they are sometimes unsharp.

AlbrechtL, 2017-08-13:
The best case is to add a custom ComboBox style to the file ["TouchComboBox.qml"](https://github.com/AlbrechtL/welle.io/blob/2b9766e241fa53ba754e5c49a039b0c10f8c2851/src/gui/QML/style/TouchComboBox.qml)

See also the ComboBox customizing documentation:
https://doc.qt.io/qt-5/qtquickcontrols2-customize.html#customizing-combobox

AlbrechtL, 2018-01-28:
Fixed in 180a9457f03550881ec522cdcaa378f0eabce8d4.

closed 2018-01-28 by AlbrechtL