AlbrechtL/welle.io #107: RTL2832u dongel not working on Windows 10.

PD1LOI, 2017-07-29:
I get welle.io not working even though I have installed the correct saddle driver See https://www.welle.io/devices/rtl_sdr

andimik, 2017-07-29:
Please run Zadig as administrator. It seems it doesn't not apply for admin rights by itself. 

Then post what messages you get.

PD1LOI, 2017-07-29:
I've all done, Sdrsharp does work.
[image]

andimik, 2017-07-29:
Do you have two interfaces?

PD1LOI, 2017-07-29:
Yes, but the dongel just works in Sdrsharp, he just does it.
[image]
[image]

AlbrechtL, 2017-07-29:
Two interfaces are not supported at the moment. See issue #95 for more details.

If you are able to recompile welle.io you can change [this line in CRTL_SDR.spp](https://github.com/AlbrechtL/welle.io/blob/7fe91a819a80f12c68b4237105422d39d1b8b269/src/input/CRTL_SDR.cpp#L107) from  
` ret = rtlsdr_open(&device, 0);` to ` ret = rtlsdr_open(&device, 1);` to use the second interface.

PD1LOI, 2017-07-29:
Strange it has always worked, but after the last version of w10 it no
longer works.

AlbrechtL, 2017-08-02:
Can you post the command line output please?

PD1LOI, 2017-08-03:
[image]

AlbrechtL, 2017-08-03:
Yes, it is issue #95. Please follow #95 for a solution.

Now I close this issue because it is duplicated.

closed 2017-08-03 by AlbrechtL

PD1LOI, 2017-08-04:
I can not recompile the file. how do you do that?