AlbrechtL/welle.io #108: usb_open error -12 with rtl-sdr v2 dongle

HansJA, 2017-08-02:
Welle.io says it detects two dongles and selects the wrong one, which isn't connected, so this fails.
I then move the dongle to the other usb port, and start welle.io again. Then it detects the dongle but returns the message "usb_open error -12". 
What's the issue? The dongle works with sdr#, sdr-j and linrad on the same windows 10 hp gl5 laptop.

I also think auto selection of a receiver is a bad idea unless one also can manually override this and force-select a receiver.

AlbrechtL, 2017-08-02:
Sorry but two ,interfaces are not supported at the moment. See issue #95 for more details.

If you are able to recompile welle.io you can change [this line in CRTL_SDR.spp](https://github.com/AlbrechtL/welle.io/blob/7fe91a819a80f12c68b4237105422d39d1b8b269/src/input/CRTL_SDR.cpp#L107) from  
` ret = rtlsdr_open(&device, 0);` to ` ret = rtlsdr_open(&device, 1);` to use the second interface.

We are thinking to add a command line option welle.io to select the right one.

closed 2017-08-02 by AlbrechtL

HansJA, 2017-08-03:
But my complaint was that welle.io can't use my SINGLE, ONE AND ONLY,
rtl-sdr dongle because I've connected it to more than one usb port.

Welle.io thinks I have two dongles, but Ihave only one.
If my dongle is actually connected to the i/f welle.io thinks is dongle 1,
it selects i/f 0, but gets no response and gives up.
If my one and only dongle is connected the other usb port, the one welle.io
thinks is i/f 0, it tries to open it but fails with "usb_open error -12".

That does not seem to be the simple problem of welle.io always selecting
interface 0.

I certainly hope the

Hans J. Albertsson
From my Nexus 5

andimik, 2017-08-03:
Could you tell us if the USB port 1 is USB 2 or 3? And what about the other one. I guess one is 2.0, the other 3.0

HansJA, 2017-08-03:
No, both ports involved are usb2.0.

Hans J. Albertsson
From my Nexus 5

HansJA, 2017-08-03:
There IS a third, usb 3.0, port, but that won't handle the usb bulk-in
drivers and isn't used.

Hans J. Albertsson
From my Nexus 5

Den 3 aug. 2017 16:32 skrev "Hans J Albertsson" <hans.j.albertsson@gmail.com

AlbrechtL, 2017-08-03:
This is some funny bug or feature of the rtl-sdr driver. Please see issue #95 for details.

HansJA, 2017-08-04:
What's the toolchain needed for compiling welle.io on w10?

Hans J. Albertsson
From my Nexus 5

Den 3 aug. 2017 9:18 em skrev "Albrecht Lohofener" <notifications@github.com