AlbrechtL/welle.io #109: Italian RAI slides not displayed - Subdirectories supported?

andimik, 2017-08-10:
Seems RAIWay defined subdirectories for their slides, for example

`Data/crit/radio1`

[image]
which prevents welle.io from being displayed at all.

See zipped eti file below (use Rai Radio 1) which you can load in welle.io by writing the rawfile in the latest odr-dabmod

 `odr-dabmod eti2raw.ini`

whereas this `eti2raw.ini` is
```
[input]
transport=file
source=rai12d.eti
loop=0

[modulator]
gainmode =1
rate=2048000

[output]
output=file

[fileoutput]
format=u8
filename=/tmp/test.raw
```
[rai12d.eti.zip](https://github.com/AlbrechtL/welle.io/files/1214670/rai12d.eti.zip)

Syntax is `welle-io -F /tmp/test.raw -D rawfile -B u8`

Hint: qt-dab displays it: 

> going to show picture Data/crit/radio1/b0038b3e-ed9a-45b6-9a6f-69363f49b3b4.jpg

basicmaster, 2017-08-10:
I just answered in the referenced DABlin issue.

AlbrechtL, 2017-08-10:
Thanks for your investigation!

closed 2017-08-13 by andimik

AlbrechtL, 2017-08-13:
@andimik Please keep an eye on dablin. If Stefan (basicmaster) added some X-PAD related features to the dablin code then it is worth to update the welle.io X-PAD decoding code.

andimik, 2017-08-13:
sure :-)

andimik, 2017-08-15:
See https://github.com/Opendigitalradio/dablin/issues/28#issuecomment-322298269

AlbrechtL, 2017-08-15:
Can you provide ready-to-go [RAW file](https://www.welle.io/devices/rawfile#record-an-raw-file) please?

andimik, 2017-08-15:
Well, it's quite easy to get a ready raw file following my howto from the 1st posting ;-)

Anyway, it's here: https://we.tl/gz5P4TeyIx

(RAI 12D Monte Venda received on Dobratsch, Austria)

AlbrechtL, 2017-08-15:
Thank for your very much courtesy (I don't have an ODR installation right now).

AlbrechtL, 2017-08-20:
In ef3b61f I added the latest X-PAD decoding code from https://github.com/Opendigitalradio/dablin/tree/37065157a2b145b70fa3df8c7a1e9486ba0faa35 with enabled loose mode.
@basicmaster Thanks for adding this option!

Now you can see the "Rai Radio1" dynamic label and slide show.
[image]