AlbrechtL/welle.io #116: Changes pulse audio channel settings

nottledim, 2017-08-14:
On Debian linux whenever welle.io is started it creates a new pulse audio channel which means it always uses default settings.  It also does the same when moving or resizing the window. 
 
It seems to be setting to defaults when audio channel is opened.

AlbrechtL, 2017-08-15:
It is a limitation of QMultimedia and the pulse back-end. If somebody has a solution they can post it here.

coldcoff, 2019-11-11:
The QAudioOutput also allows to be constructed with a given QAudioDeviceInfo as target, see:  https://doc.qt.io/qt-5/qaudiooutput.html and https://doc.qt.io/qt-5/qaudiodeviceinfo.html.

-> So you could maybe alternatively query the available devices and save an own default device in welle.io's preferences. Then the magically created stream name is irrelevant.

Would this be possible, please? I think it's quite common to for example point welle.io's sound to a headphone, instead of the default notebook device (which is the builtin speaker).

AlbrechtL, 2019-11-24:
Thanks for pointing this out! It is worth to try it.

If somebody can send a pull request this would be great!

AlbrechtL, 2019-11-26:
I looked into it. It is not as easy.
In my case `QAudioDeviceInfo` gives me plenty of audio devices but I have only one connected.
Also the names are quite strange. I don't think that these names are really helping of finding the correct audio output. 
```
2019-11-26T20:15:33.717 Debug: Audio: Available sound output device:  "default"
2019-11-26T20:15:33.717 Debug: Audio: Available sound output device:  "jack"
2019-11-26T20:15:33.717 Debug: Audio: Available sound output device:  "pulse"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "sysdefault:CARD=PCH"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "front:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "surround21:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "surround40:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "surround41:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "surround50:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "surround51:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "surround71:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dmix:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dsnoop:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hw:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "plughw:CARD=PCH,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "usbstream:CARD=PCH"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hdmi:CARD=NVidia,DEV=0"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hdmi:CARD=NVidia,DEV=1"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hdmi:CARD=NVidia,DEV=2"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hdmi:CARD=NVidia,DEV=3"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dmix:CARD=NVidia,DEV=3"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dmix:CARD=NVidia,DEV=7"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dmix:CARD=NVidia,DEV=8"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dmix:CARD=NVidia,DEV=9"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dsnoop:CARD=NVidia,DEV=3"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dsnoop:CARD=NVidia,DEV=7"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dsnoop:CARD=NVidia,DEV=8"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "dsnoop:CARD=NVidia,DEV=9"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hw:CARD=NVidia,DEV=3"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hw:CARD=NVidia,DEV=7"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hw:CARD=NVidia,DEV=8"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "hw:CARD=NVidia,DEV=9"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "plughw:CARD=NVidia,DEV=3"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "plughw:CARD=NVidia,DEV=7"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "plughw:CARD=NVidia,DEV=8"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "plughw:CARD=NVidia,DEV=9"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "usbstream:CARD=NVidia"
2019-11-26T20:15:33.718 Debug: Audio: Available sound output device:  "alsa_output.pci-0000_00_1b.0.analog-stereo" 
```

BTW: The missing API for setting the pulse name is documented here: https://bugreports.qt.io/browse/QTBUG-32842

AlbrechtL, 2022-12-16:
We should take a look into the new Qt6 multimedia classes.

coldcoff, 2022-12-19:
Yes, that would be a wonderful christmas present ;-)

AlbrechtL, 2026-10-22:
Closed because the system audio output selector can be used.

closed 2026-10-22 by AlbrechtL