AlbrechtL/welle.io #117: No sound with Hifiberry-DAC on Pi3

Vitalic66, 2017-08-18:
Hi,

I'm using Viktorgino's AA Headunit with welle.io integration. When starting the app I get this error message '"Audio format \"audio/pcm\" 16-bit stereo not supported. Something went wrong, cannot play audio."'. I'm using Pi3 with latest raspbian stretch and  Hifiberry-DAC for sound output. How can I fix the problem? Welle.io is running fine. I can scan for stations and get 3 green blocks when trying to listen to one. I can also see which song is played. So welle.io is working.

AlbrechtL, 2017-08-20:
welle.io uses the default audio output. Please check if your Hifiberry-DAC is configured as default audio device.

Vitalic66, 2017-08-20:
Yes it is configured as default. Does welle.io try to set up anything at start or does it only check the hard/software parameters of the output device?

AlbrechtL, 2017-08-20:
In ["CAudio.cpp" line 77](https://github.com/AlbrechtL/welle.io/blob/ef3b61f5b3f295237aa56def26b2790e24d49391/src/output/CAudio.cpp#L77) welle.io checks if the audio device supports the format. You can try to comment out this line.

Since welle.io uses [QAudioOutput](https://doc.qt.io/qt-5/qaudiooutput.html) you can also ask the QT community.

Vitalic66, 2017-08-20:
Tried commenting out the whole if rule
´´´
#    QAudioDeviceInfo info(QAudioDeviceInfo::defaultOutputDevice());
#    if (!info.isFormatSupported(AudioFormat)) {
#        qDebug() << "Audio:"
#                 << "Audio format \"audio/pcm\" 16-bit stereo not supported. Something went wrong, cannot play audio.";
#        return;
#}
´´´
Right now compiling fails:

´´´
welle.io/src/output/CAudio.cpp:77:15: error: token "." is not valid in preprocessor expressions
 #    if (!info.isFormatSupported(AudioFormat)) {
               ^
welle.io/src/output/CAudio.cpp:77:0: error: unterminated #if
 #    if (!info.isFormatSupported(AudioFormat)) {

welle.io/src/output/CAudio.cpp: In member function ‘void CAudio::init(int)’:
welle.io/src/output/CAudio.cpp:76:67: error: expected ‘}’ at end of input
     QAudioDeviceInfo info(QAudioDeviceInfo::defaultOutputDevice());
                                                                   ^
´´´

Vitalic66, 2017-08-20:
Thx for the idea. Now it works.

AlbrechtL, 2017-08-21:
Interesting, QMultimedia says that the audio format is not supported but it works.

In 4bc4091 I changed the code that the initialization is still continued even if the audio format is detected as not supported. This should help in the cases like yours.

closed 2017-08-21 by AlbrechtL