AlbrechtL/welle.io #118: Failes to compile with CMake

neural75, 2017-08-18:
Many errors appears on compiling viterbi_spiral.c
Compiles ok setting:
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -msse4.1")
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -msse4.1")
In CMakeFIles.txt

--------------------------------------------------------------------------------
 In file included from /home/trubini/src/welle.io/src/backend/viterbi_spiral.c:23:0:
/usr/lib/gcc/i686-linux-gnu/6/include/emmintrin.h:975:1: error: inlining failed in call to always_inline ‘_mm_unpackhi_epi32’: target specific option mismatch
 _mm_unpackhi_epi32 (__m128i __A, __m128i __B)
 ^~~~~~~~~~~~~~~~~~
/home/trubini/src/welle.io/src/backend/viterbi_spiral.c:697:14: note: called from here
         s225 = _mm_unpackhi_epi32(s221, s222);
         ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/trubini/src/welle.io/src/backend/viterbi_spiral.c:23:0:
/usr/lib/gcc/i686-linux-gnu/6/include/emmintrin.h:999:1: error: inlining failed in call to always_inline ‘_mm_unpacklo_epi32’: target specific option mismatch
 _mm_unpacklo_epi32 (__m128i __A, __m128i __B)
 ^~~~~~~~~~~~~~~~~~
/home/trubini/src/welle.io/src/backend/viterbi_spiral.c:696:14: note: called from here
         s224 = _mm_unpacklo_epi32(s221, s222);
         ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[cut]

AlbrechtL, 2017-08-20:
Which OS (in case of Linux which distribution) do you use?

AlbrechtL, 2017-09-19:
Closed because there are no more information to analyse the issue.

closed 2017-09-19 by AlbrechtL