AlbrechtL/welle.io #119: RTL2838UHIDIR running on Windows, but not on Android

joachimkr, 2017-08-18:
Hi,
I have a new device and it is working with Windows. 'found Rafael R820 tuner' is reported in windows and it is working smoothly.
Initial intent was to run it on an Android, so this was just a counter-proof.
On Android, well.io reports, it doesn't find a tuner.
My test device is a Samsung A5-2016 running Samsungs installation of Android 6.
Crashes appeared, the device rebooted some times during tests.

Questions:
1. Is it required to root the device? If so, why?
2. My REAL target is a car DVD Android 6, running on a Intel quadcore.
How can I ever run this on that device?
Thanks for help,
Joachim

joachimkr, 2017-08-18:
Correction. The testing device Galaxy A5 (2016) SM-A510F is running Android 7.0, Kernel 3.10.61 from july 3

AlbrechtL, 2017-08-20:
Does the [RTL2832U driver](https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro) reports any messages if you run the app "Rtl-sdr driver" in advanced mode?

To Q1: No, it is not necessary to root your device.
To Q2: At the moment welle.io is only available for Android with armv7 CPU architecture. It should be possible to compile welle.io also for x86 Android. But we never tried it. If you like you can try it (https://github.com/AlbrechtL/welle.io#android)

joachimkr, 2017-08-20:
Thanks AlbrechtL, How can I enter the advanced mode of the sdr driver app?

I will try a make for an Android x86 and let you know progress here. Unfortunately, I do not know, how to get error messages/logs during start.
If I do a make, what about the driver app? Is the driver app part of the Welle.io repository?
If not, where can I find that app's sources?

AlbrechtL, 2017-08-20:
welle.io uses the ["RTL2832U driver"](https://play.google.com/store/apps/details?id=marto.rtl_tcp_andro)([sources](https://github.com/martinmarinov/rtl_tcp_andro-)) from Martin Marinov.

joachimkr, 2017-08-22:
I found the driver app's advanced mode. On the Samsung A5 (2016) this mode does not give bad messages.
Welle.io crashes on startup, when starting driver load. Most times the device reboots after a minute. 
I used an additional powerplug, in order to avoid undervoltage conditions.
Is there a log file?
I understood, driver and application use a tcp protocol. Is that legal in Android 7?

AlbrechtL, 2017-08-22:
Yes, to use TCP is OK. I have Android 7.1.2 and it is working perfectly.

Maybe there is a driver issue?
@martinmarinov Do you have an idea how to analyze this problem?

joachimkr, 2017-08-22:
Today I used another device (Tablet Samsung)
and have also no success to run welle.io

It reports, it cannot access the port 1234 on local host.

Here is the output of the driver app screen:

onStart
RtlSdr: found 1 device opening options
Only 1 option available, no need to ask user. Opening rtl-sdr
/dev/bus/usb/002/006
Queueing
Arguments SdrTcpArguments{gain=24, samplerateHz=1024000,
frequencyHz=91800000, address='0.0.0.0', port=14423, ppm=0}
Starting service with device rtl-sdr /dev/bus/usb/002/006
Opening fd 57
Set sampling rate to 1024000
Tuner gain set to 2.400000 dB
SdrTcp: Listening on 0.0.0.0:14423
Device was open. Closing the prompt activity.
SdrTcp: Waiting for client...
The rtl-tcp implementation is running and is ready to accept clients
Acquired wake lock. Will keep the screen on.
Starting was successful!
rtlsdr_read_async failed
rtlsdr_read_async finished successfully
SdrTcp: Requested sdrtcp stop asynchroneously
Closed service due to exception SdrException: Unknown error!
Wake lock released
Closing service
SdrTcp: TCP server shutting down.
SdrTcp: Closing sdrtcp due to main thread finishing
SdrTcp: Closing from state 0
SdrTcp: Server thread shut down
UsbDelegate USB attached: /dev/bus/usb/002/008
UsbDelegate no activity found for SDR handling

martinmarinov, 2017-08-23:
Hey @joachimkr. Can you try running other free SDR software on Android (such as the demo of SDR Touch) on your devices? This will confirm whether the problem is in the driver.

joachimkr, 2017-08-24:
It has to do with Android and device.
I tried 2 Android devices. Samsung Tab 3 with Android 6 and Samsung A5 2016
with Android 7.
None is rooted.
Both show with SDR Touch and Welle.io a very similar error / behaviour.
Here is, what happens with SDR Touch.
On start,  they start loading the driver.
Android 7 keeps to show loading, until sudden death, reboot.
Android 6 shows loading, then shows a white popup, unexpected unplug,
'unerwartet getrennt' or 'Fehler beim Befehl senden'.

Conditions are the same, the device is a
NooElec R820T SDR & DVB-T
NESDR Mini

Power is coming from the USB OTG Adapter.
I also tried direct powering, same results.

The USB Receiver device is working smoothly on Windows 10, even across a
USB hub.

If you have further proposals for applications, I will give it a try. If
you want the receiver device for debugging, you will get it.

Regards, thanks for help
Joachim

AlbrechtL, 2017-08-25:
You can give [NooElec NESDR Nano 3](https://www.nooelec.com/store/nesdr-nano-three-otg.html) a try.

You find it also on Amazon Germany: https://www.amazon.de/NooElec-NESDR-Nano-Aluminium-Geh%C3%A4use-R820T2-basierte/dp/B073JZ8CC2

joachimkr, 2017-09-03:
> NESDR-Nano-Aluminium-Geh%C3%A4use-R820T2-basierte/dp/B073JZ8CC2

Hi Albrecht,

Thanks for your help!
I ordered that nano device immediatly and gave that a try, today!
Because I started at my smartphone, the trial failed...
Seems, my  OTG cable is the problem - at least!

By trying many device combinations I assume, my OTG cable is now entirely
dead. Before, the old, the cheap receiver was probably underpowered. But it
gave a response to the driver through the OTG cable.
I don't know how this OTG cable adapter did that, but it damaged the nano
immediatly at first connection and now cable and nano are dead, both.

I checked in a USB hub with the PC and found, welle.io doesn't give any
spectrum signal for the nano, while it is more then hot after a minute.
Antenna is the same. Or do I need another driver?

Additional observation is, that the power connection of the (USB3) hub is
much too noisy, when loaded by current-hungry devices. I am sure, this can
be fixed with cables, which are buffered with capacitors and chokes.
The cheap device has useful reception only, when I disconnect any other
device from the hub.

By the way, a proposal:
Automatic re-tune of a station occurs, from time to time, which at first
disrupts stream from it - silence for some seconds.
I propose, to introduce a expert switch, disabling that retune.
And, to improve that overall reception, add the capability to drive a
second tuner, to switch over, in case of better signal. signal/noise ratio.
Keep processing power low...
I want to use all that in a car, a double tuner is good, required anyhow,
then.

I am glad with welle.io,
just DAB, with best reception.

Regards
Joachim

joachimkr, 2017-09-05:
Hello again,

just to share my progress:
Today, I got another OTG adapter.
It works with the cheap NooElec NESDR mini.
On both devices, Samsung Tab 3 and Samsung Galaxy A5 2016.

That damn simple OTG cable was the troublemaker.
I guess, there was a FET switch builtin, which was intended to make it
simple to use.
Now, I have a device with 3 USB A and 1 USB C socket. It offers a mechanic
switch, so I give external power.

The NooElec runs perfect with Welle.io with this cable!
It seems, this device is very sensitive to powerline noise. And this OTG
adapter is a reasonable filter.

Next I will try, to run the driver on the Intel platform, car Android
radio, which is the real target.

I hope, you will give me help to run Welle.io on that ;-)

Greetings,  best regards
Joachim

closed 2017-09-06 by joachimkr