AlbrechtL/welle.io #120: Version? / Font Blurry

nousername1, 2017-08-19:
1) I miss some Version Info.
2) On Windows 7 the Font is Blurry and show some Wobbling thing.

I believe its the Version from June?! It its still fix sry.

andimik, 2017-08-19:
Have you clicked on ?

AlbrechtL, 2017-08-20:
To Q1: You can find the version info if you click inside the settings view to the top right question mark.
[image]

To Q2: Please update your graphic card driver.

nousername1, 2017-08-20:
So I got the current Version. The ? show just in the Settings Menu....
And I use the current Driver for my Thinkpad. In a old Version the Font was normal not blurry.

AlbrechtL, 2017-08-20:
Please post the command line output.

AlbrechtL, 2017-09-19:
Closed because there are no more information to analyse the issue.

closed 2017-09-19 by AlbrechtL