AlbrechtL/welle.io #122: FM Radio

CodeHub00, 2017-08-22:
Rtl-Sdr supports Fm Radio signals.
So maybe welle.io can receive analog Fm radio too.

viktorgino, 2017-08-22:
@AlbrechtL I have an FM radio GUI as part of my project ([viktorgino/headunit-desktop](https://github.com/viktorgino/headunit-desktop)) which I'd be happy to contribute to this project.

Qt has a radio module, which I think relies on V4L. Also there are some other projects focused on SDR based FM radio such as:
http://osmocom.org/projects/sdr/wiki/rtl-sdr
https://github.com/jorisvr/SoftFM

Screenshot:
[image]

CodeHub00, 2017-08-22:
I want to use DAB+ and FM radio on my Android head unit.
On my head unit, FM radio software is useless.
So maybe later welle.io support both of them.

viktorgino, 2017-08-22:
@AlbrechtL How does rtl_fm_streamer compare (performance and quality wise) to the projects you mentioned? I'm happy to give this a shot, because its something I would really like to have as part of headunit-desktop

AlbrechtL, 2017-08-23:
I never bench-marked rtl_fm_streamer against the other solutions.

Since all devices are SDRs it doesn't matter which sticks is used. At the moment welle.io supports rtl-sdr and airspy as well as rtl_tcp and RAW input.

renne, 2017-11-01:
I suggest to add RDS-TP and RDS-TA support for FM-fallback with automatic frequency switching in car usage.

The current GUI elements are too small for car usage.

AlbrechtL, 2017-11-01:
Regrading the small GUI elements: Please try to use Android Auto https://github.com/AlbrechtL/welle.io/issues/103.

AlbrechtL, 2017-11-24:
I'm closing this issue for now. But it can be used to discuss this topic.

closed 2017-11-24 by AlbrechtL

alexmyczko, 2019-02-20:
Ohhhh cool! Also add Airbands?