AlbrechtL/welle.io #127: New welle.io forum (https://forum.welle.io)

AlbrechtL, 2017-09-07:
Today I added a forum https://forum.welle.io/ for this project.

The goal is to provide also non-English speakers the opportunity to get help in their native language and to have a little bit more flexible discussions as it is possible with issues.

At the moment some German and English categories are available.

If you need more categories and languages just let me know!

closed 2017-10-18 by AlbrechtL

viktorgino, 2017-10-18:
Hi Albert

I can try to provide help in Hungarian.

In case you want to do this here are the translations for the forum topics:
Hungarian -> Magyar
General -> Általános
General discussions -> Általános társalgó
Support -> Támogatás
Installation, Issues, Devices -> Telepítés, Problémák, Eszközök
Showcase -> Showcase
Show us your welle.io set ups and applications -> Mutasd be hogyan használod a welle.io-t
Development -> Fejlesztés
Feature Requests, Bugs -> Fejlesztési igenyek, Programhibák

AlbrechtL, 2017-10-19:
Thanks!
Now, there are also Hungarian categories available: https://forum.welle.io/categories/magyar_hu