AlbrechtL/welle.io #131: Request to kill rtl-sdr driver after close the app

Tibor69, 2017-09-18:
The first launch of the app is OK, but I close it and I relaunch the app it freezed, because it can't load the rtl-sdr driver. After I closed app the rtl-sdr driver still loading. 
When I kill the rtl-sdr driver manually, after the app launch good. 
Can you make that when the app close the rtl-sdr driver will be killed?

AlbrechtL, 2017-09-18:
@twoofseven Does the upcoming version improve this?

twoofseven, 2017-09-18:
The latest version (pull request #128) solves some issues, especially for the background service to recognise RTD-SDR disconnects and handling of reconnects.
From the bug description I am not really sure, whether I understand the use case, because closing the welle.io app e.g. via "Exit welle.io" within the settings screen with the version before #128 should have had no influence on the background service. With #128, the RTL-SDR connection will be closed upon "Exit welle.io" within the app or the close button within the notification. Relaunching might break, due to the fact that several components and different processes are involved to succeed (background service, Qt environment, and RTL-SDR driver startup). 
Therefore a logcat trace would help a lot to identify the issue.

Tibor69, 2017-09-19:
I don't know logcat trace, therefor I made a video from issue. 
You can access from this link:
https://mega.nz/#F!chYk0JQb!_F0n2U52H6b3E5PbZTxDVg

twoofseven, 2017-09-19:
@Tibor69 Thanks for the video. That helped to understand the issue.
Which version of the app are you currently using? The official release from Play Store or any of the version in development of #103 (e.g. 20170917_Android_armv7_welle-io.apk)?
Because within the video I do not see a welle.io notification, which will be shown if the new welle.io app version is active.
Beside that I do not think that the new version will fix the issue, because it seems, that during the second attempt the welle.io app launched  (via Android intent) the RTL-SDR driver initialisation , which got stuck. 
But we could try to explicitly send the following TCP command:
`TCP_ANDROID_EXIT = 0x7e, // send any value to cause the driver to turn off itself`
as documented in https://github.com/martinmarinov/rtl_tcp_andro-/blob/master/app/src/main/jni/tcp_commands.h
before closing the TCP connection.
Could you please try to reproduce the issue with the new version? If the issue still exist, I can prepare the fix and you could verify it again.

Tibor69, 2017-09-20:
I'm using the official version: 1.0-beta.
I tried installation of the 20170917_Android_armv7_welle-io.apk on my mobile and car head unit, but I got this message while installation: 
Parse error / There was a problem while parsing the package. 
My head unit details:
RK3066 chip Dualcore 1,6GHz CPU. 800x480 dot display, Kitkat 4.4.4 rooted OS.

twoofseven, 2017-09-20:
I added the Android RTL-SDR driver's TCP_ANDROID_EXIT cmd support in #135

closed 2017-11-24 by AlbrechtL