AlbrechtL/welle.io #134: Hungarian translation

AlbrechtL, 2017-09-20:
Thanks to Tibor Kovács there is a hungarian translation in progress.

[image]

A first development version can be download here: [20170920_welle-io-setup.exe](https://gist.github.com/AlbrechtL/1fb6cc890863b21fe524b9e9ea9766d0/raw/fcd9875ed4fd7e5f5a3848aaddd138dd360828f2/20170920_welle-io-setup.exe)

@Tibor69 Please verify the translation and test if the auto detecting of hungarian language is working correctly.

Tibor69, 2017-09-21:
Hi,

I have verified the localization. I repaired the ts file. It will be OK
now. I attached it.

I have a request:
Can you set to 6 bar of the signal strenght signer from 3 bar?

Bye
Tibor

AlbrechtL, 2017-10-04:
Your updates are included in https://github.com/AlbrechtL/welle.io/commit/42971b17da2c4c879ce42ab330b78ef1c05dffcb.

closed 2017-10-04 by AlbrechtL