AlbrechtL/welle.io #136: Windows7-install missing components?

sm3ulc, 2017-09-21:
Installed on a fresh Win7. Some graphical reqs missing?

[image]

andimik, 2017-09-21:
Have you tried the default win 7 skin?

sm3ulc, 2017-09-21:
Yes. Same result.

AlbrechtL, 2017-09-21:
I assume some graphic card issues because welle.io is using OpenGL for the GUI. Please try to update your graphic card driver.

sm3ulc, 2017-09-26:
The Windows7 is running in vmware workstation. I've checked and it reports OpenGL 2.1 and documents claim 3.3.

AlbrechtL, 2017-10-04:
Your set up is really special.

Officially we don't support Windows 7 nevertheless it is working in the most cases. We tested welle.io in a [VirtualBox](https://www.virtualbox.org/) with [this](https://developer.microsoft.com/en-us/windows/downloads/virtual-machines) Windows 10 image.

We strongly recommend to switch to Windows 10.

closed 2017-10-04 by AlbrechtL