AlbrechtL/welle.io #138: SDRPlay support is missing

dmitryelj, 2017-09-23:
Hi.

In your project soapysdr devices are promised to be supported, but I tried to use SDRPlay without success.
welle-io.exe -D "driver=sdrplay"
welle-io.exe -D "sdrplay"

Result: "Error while opening device"

**SoapySDRUtil.exe --probe="driver=sdrplay"** device info:

Probe device driver=sdrplay
Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_003.010.002.000-0-gbd6e21dc

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=2 1603017600
  mir_sdr_api_version=2.100000
  mir_sdr_hw_version=2

  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: Antenna A, Antenna B, Hi-Z
  Corrections: DC removal
  Full gain range: [0, 47] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 8] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

AlbrechtL, 2017-09-23:
soapysdr devices are only available under Linux. It seems that you use Windows.

dmitryelj, 2017-09-23:
Thanks for the answer. Soapy is a cross-platform driver, it works on Windows as well. When possible, please check whats wrong.

AlbrechtL, 2017-09-24:
The issue is not the Soapy driver the problem is that the Soapy input is only available in the Linux version of welle.io. If you are a developer feel free the port it to Windows.

I also added it to the [open task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

closed 2017-10-04 by AlbrechtL