AlbrechtL/welle.io #141: Create a Homebrew/Linuxbrew formula

russel, 2017-10-05:
Particularly for macOS (and those of us stuck on OSX) it might be easier to use Homebrew to install Welle.Io. This requires creating and adding a formula to the Homebrew database.

This would also allow Welle.Io to be accessible via Linuxbrew.

Perhaps also need to submit a proposal to the MacPorts folk to have Welle.Io supported there.

muellermartin, 2017-10-06:
You can use this formula and submit it to Homebrew if you like: https://github.com/muellermartin/homebrew-distill/blob/master/welle.io.rb

AlbrechtL, 2017-10-07:
I'm open to support Homebrew. Since I don't have a mac I'm not able to maintain the formula. Does somebody would like to do this job?

russel, 2017-10-07:
This is not just restricted to macOS (Homebrew) but can also be used (albeit in a slightly altered form) for Linuxbrew. I think changing the title to seem to restrict to macOS should either be altered or there should be a [Linux] equivalent made.

russel, 2017-10-07:
If @muellermartin isn't up for formally submitting his version, then the way forward would seem to be to create a new repository just for the Homebrew and Linuxbrew formulas and then copy Martin's formula to it. Whilst I have an OSX machine, it is not one I use for anything other than testing builds. So I could be involved in a Homebrew formula to get it going, and maintain it in the short-term, but it would be with the view to having someone else take up maintenance long-term. Obviously I have Linux machines a plenty, so can be involved in the Linuxbrew as well.

AlbrechtL, 2018-01-22:
I added the formula in b05474b.
Can someone provide a description how to use it (@muellermartin or @christophmuellerorg)?

muellermartin, 2018-01-23:
Thanks for adding the formula!

I'd be happy to provide a section in the README and/or Wiki. There is one needed change though: If you want to include the formula in this repository, according to the [official Homebrew guide for taps](https://github.com/Homebrew/brew/blob/master/docs/How-to-Create-and-Maintain-a-Tap.md#creating-a-tap) you need to name the directory containing the formulas either `Formula` or `HomebrewFormula` instead of the current `macOS`.

After that it basically boils down to `brew install AlbrechtL/welle.io/welle.io` which automatically installs the tap and the Formula named `welle.io`.

AlbrechtL, 2018-01-23:
You are welcome to send a pull request with any changes that are necessary. A README.md entry would be great! You can also update the website via github. The side is hosted in the branch ["gh-pages"](https://github.com/AlbrechtL/welle.io/tree/gh-pages), see also the [wiki](https://github.com/AlbrechtL/welle.io/wiki/Website).

sds89ufdg, 2018-02-07:
For anyone who wants to test the forumale before the update:

    brew install cmake qt fftw faad2 librtlsdr libusb
    git clone https://github.com/AlbrechtL/welle.io.git
    cd welle.io/macOS
    brew install -f --HEAD ./welle.io.rb

run:
    /usr/local/Cellar/welle.io/HEAD-de9b1cb/bin/welle-io

I cannot confirm if your HEAD-xxxxxxxx directory will be the same but you'll see it easily enough.

It'll be interesting whether this will work via LinuxBrew as well, once the formuale is available.

pr2git, 2018-02-14:
Hi,

welle.io doesn't compile on OSX 10.13.3, so homebrew doesn't build neither:

```
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:180:13: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
            PaUtil_WriteMemoryBarrier();
            ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:88:40: note: expanded
      from macro 'PaUtil_WriteMemoryBarrier'
#   define PaUtil_WriteMemoryBarrier() OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
      been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:189:13: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
            PaUtil_FullMemoryBarrier();
            ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:86:40: note: expanded
      from macro 'PaUtil_FullMemoryBarrier'
#   define PaUtil_FullMemoryBarrier()  OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
      been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:226:17: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
                PaUtil_FullMemoryBarrier(); /* (write-after-read) => ful...
                ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:86:40: note: expanded
      from macro 'PaUtil_FullMemoryBarrier'
#   define PaUtil_FullMemoryBarrier()  OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
      been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:264:17: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
                PaUtil_ReadMemoryBarrier(); /* (read-after-read) => rea...
                ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:87:40: note: expanded
      from macro 'PaUtil_ReadMemoryBarrier'
#   define PaUtil_ReadMemoryBarrier()  OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
[... 51 lines cut ...]
    const std::array<float, 1>  fc2580_gains {{ 0 /* no gain values */ }};
                                ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/__tuple:222:64: note: 
      template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:30:
/Users/pr2/Github/welle.io/src/input/CRTL_TCP_Client.h:121:34: error: 
      implicit instantiation of undefined template 'std::__1::array<float, 29>'
    const std::array<float, 29>  r82xx_gains {{ 0.0, 9.0, 1.4, 2.7, 3.7...
                                 ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/__tuple:222:64: note: 
      template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
/Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:125:27: error: 
      assigning to 'CVirtualInput *' from incompatible type 'CRTL_TCP_Client *'
            InputDevice = new CRTL_TCP_Client(RadioController);
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
5 warnings and 6 errors generated.
make[2]: *** [CMakeFiles/welle-io.dir/src/input/CInputFactory.cpp.o] Error 1
make[1]: *** [CMakeFiles/welle-io.dir/all] Error 2
make: *** [all] Error 2

```

muellermartin, 2018-02-14:
@pr2git This looks like it might be related to PulseAudio. Maybe it helps if you install it via `brew install pulseaudio`... If it works, then it is a missing dependency.

pr2git, 2018-02-14:
I installed pulseaudio as advised and here is the result now

```
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:180:13: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
            PaUtil_WriteMemoryBarrier();
            ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:88:40: note: expanded
      from macro 'PaUtil_WriteMemoryBarrier'
#   define PaUtil_WriteMemoryBarrier() OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
      been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:189:13: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
            PaUtil_FullMemoryBarrier();
            ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:86:40: note: expanded
      from macro 'PaUtil_FullMemoryBarrier'
#   define PaUtil_FullMemoryBarrier()  OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
      been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:226:17: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
                PaUtil_FullMemoryBarrier(); /* (write-after-read) => ful...
                ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:86:40: note: expanded
      from macro 'PaUtil_FullMemoryBarrier'
#   define PaUtil_FullMemoryBarrier()  OSMemoryBarrier()
                                       ^
/usr/include/libkern/OSAtomicDeprecated.h:749:9: note: 'OSMemoryBarrier' has
      been explicitly marked deprecated here
void    OSMemoryBarrier( void );
        ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:28:
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.h:33:
In file included from /Users/pr2/Github/welle.io/src/CRadioController.h:39:
In file included from /Users/pr2/Github/welle.io/src/output/CAudio.h:34:
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:264:17: warning: 
      'OSMemoryBarrier' is deprecated: first deprecated in macOS 10.12 - Use
      std::atomic_thread_fence() from <atomic> instead
      [-Wdeprecated-declarations]
                PaUtil_ReadMemoryBarrier(); /* (read-after-read) => rea...
                ^
/Users/pr2/Github/welle.io/src/various/ringbuffer.h:87:40: note: expanded
      from macro 'PaUtil_ReadMemoryBarrier'
#   define PaUtil_ReadMemoryBarrier()  OSMemoryBarrier()
[... 54 lines cut ...]
                                ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/__tuple:222:64: note: 
      template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
In file included from /Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:30:
/Users/pr2/Github/welle.io/src/input/CRTL_TCP_Client.h:121:34: error: 
      implicit instantiation of undefined template 'std::__1::array<float, 29>'
    const std::array<float, 29>  r82xx_gains {{ 0.0, 9.0, 1.4, 2.7, 3.7...
                                 ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/__tuple:222:64: note: 
      template is declared here
template <class _Tp, size_t _Size> struct _LIBCPP_TEMPLATE_VIS array;
                                                               ^
/Users/pr2/Github/welle.io/src/input/CInputFactory.cpp:125:27: error: 
      assigning to 'CVirtualInput *' from incompatible type 'CRTL_TCP_Client *'
            InputDevice = new CRTL_TCP_Client(RadioController);
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
5 warnings and 6 errors generated.
make[2]: *** [CMakeFiles/welle-io.dir/src/input/CInputFactory.cpp.o] Error 1
make[1]: *** [CMakeFiles/welle-io.dir/all] Error 2
make: *** [all] Error 2
```
Seems that it doesn't improve compilation and the error are the same.
Since there are some deprecated instructions used, they need to be changed in the code source.

muellermartin, 2018-02-14:
Hm, that did not change anything. Seems that I was mistaken by the first `PaUtil_` prefix. I think the relavant part is `error: implicit instantiation of undefined template 'std::__1::array<float, 14>' const std::array<float, 14> e4k_gains...` but I have no clue what causes this error...

AlbrechtL, 2018-02-14:
Can it be related with https://github.com/AlbrechtL/welle.io/commit/5708bfad6ea1e2fadc5d1d9db39f99422302cc0f#commitcomment-27552913?

pr2git, 2018-02-15:
Solution to have compilation working is

```
diff --git a/src/input/CRTL_TCP_Client.h b/src/input/CRTL_TCP_Client.h
index 32a616b..e109a66 100644
--- a/src/input/CRTL_TCP_Client.h
+++ b/src/input/CRTL_TCP_Client.h
@@ -36,6 +36,7 @@
 #include <QTcpSocket>
 #include <QTimer>
 #include <QtNetwork>
+#include <array>
 
 #include "CVirtualInput.h"
 #include "DabConstants.h"
```

muellermartin, 2018-02-15:
@pr2git Confirmed. Also works for me.

muellermartin, 2018-02-16:
I've created PR #204 to improve the compatibility with Homebrew.

With this patch welle.io can be installed with two commands: `brew tap AlbrechtL/welleio https://github.com/AlbrechtL/welle.io` followed by `brew install AlbrechtL/welleio/welle.io`.

*Note*: The tap command actually only should be installed via `brew tap AlbrechtL/welle.io` but apparently Homebrew fails to do so when the name contains a dot (this was already reported as bug, but likely will not get fixed: Homebrew/brew#3811).

I will also try to update the README.

pr2git, 2018-02-16:
You escape the . like this:  brew tap AlbrechtL/welle\.io  but anyway it search for: albrechtl/homebrew-welle.io so anyway the homebrew- prefix is added.So I don't think that the dot is the only problem.

muellermartin, 2018-02-16:
@pr2git You're right, actually `brew tap AlbrechtL/welle.io` would not be enough because of the automatically added prefix, while `brew tap AlbrechtL/welle.io https://github.com/AlbrechtL/welle.io` should work (but does not due to the bug). Actually tapping does work but installing any formula from a tap containing a dot fails. Escaping the dot with a backslash does not change anything for me.

AlbrechtL, 2018-02-16:
Is it possible to write "welle_io" instead of "welleio". I assume that a lot of people will add a "." by mistake.

muellermartin, 2018-02-16:
Hm, good question but I think a underscore should also work.

AlbrechtL, 2018-02-17:
Can you add this to the pull request?

muellermartin, 2018-02-17:
Sure! Already done ;)

AlbrechtL, 2018-02-17:
Merged.

closed 2018-02-17 by AlbrechtL