AlbrechtL/welle.io #146: [Android, RTL_TCP] Support also other tuners than R820T/T2

AlbrechtL, 2017-10-09:
In the RTL_TCP input driver the gains are [hard coded](https://github.com/AlbrechtL/welle.io/blob/d5e99814528f94ab71e02d8f36410673634bc751/src/input/CRTL_TCP_Client.cpp#L372) because the rtl_tcp protocol doesn't implement a function to read out the existing gains.

Because the most rtl-sdr dongles are using the R820T/T2 tuner these gains are used. But if a user tries to use a dongle with an other tuner e.g. "Fitipower FC0012" the driver still uses the R820 gains which results in wrong gain settings.
 
The following example shows the "Fitipower FC0012" gains.
   ```
RTL_SDR: Open rtl-sdr
RTL_SDR: Found 1 devices. Uses the first working one
Found Fitipower FC0012 tuner
RTL_SDR: Opening rtl-sdr device 0
RTL_SDR: Supported gain values 5
RTL_SDR: gain 19.2
RTL_SDR: gain 17.9
RTL_SDR: gain 7.1
RTL_SDR: gain -4
RTL_SDR: gain -9.9
   ```

This issue is to discuss possible solutions.

zvpunry, 2017-11-01:
I think that "-D soapysdr --sdr-driver-args driver=remote,remote=192.168.1.42,remote:driver=rtlsdr" could be a temporary workaround for this bug.

SoapySDR, SoapyRTLSDR and SoapyRemote must be installed on the system that would normally runs rtl_tcp. SoapySDR and SoapyRemote must be installed on the system running welle-io and it must be compiled with "cmake .. -DSOAPYSDR=1" to make this work.

fvwcldgm, 2018-01-29:
What are the values for R828D2 ?

gvanem, 2018-01-29:
> In the RTL_TCP input driver the gains are hard coded ...

In the `CRTL_TCP_Client::TCPConnectionWatchDogTimeout()`, the tuner-type and gain-counts could be read (the `rtl_tcp` server sends this). To fix the problem, one could just use the same hard-coded table as `librtlrdr.c` uses in function `rtlsdr_get_tuner_gains()`.  Or no?

AlbrechtL, 2018-01-29:
Does `rtl_tcp` send the tuner type?

gvanem, 2018-01-29:
I've not verified it, but from the `listen()`-loop in `rtl_tcp.c`, it looks like it does. It sends the value of `rtlsdr_get_tuner_type()` as `dongle_info.tuner_type` (an `enum rtlsdr_tuner`).

AlbrechtL, 2018-01-29:
Cool I didn't realize it! Can you verify if we can use this?
If yes this would be a great improvement for the rtl_tcp interface which is used by default for Android.

AlbrechtL, 2018-02-13:
Fixed in 5708bfad6ea1e2fadc5d1d9db39f99422302cc0f.

closed 2018-02-13 by AlbrechtL