AlbrechtL/welle.io #151: [Discussion] SoapySDR as standard SDR device library

AlbrechtL, 2017-10-25:
[SoapySDR](https://github.com/pothosware/SoapySDR/wiki) is a SDR support library with support for a lot of SDR devices.

I'm reflecting about to replace the existing SDR device code (rtlsdr, airspy) with SoapySDR. The big advantage is that SoapySDR supports a lot of SDR devices. And we have reports that users used the existing experimental welle.io SoapySDR code with HackRF, airplay and USRP successfully.

AlbrechtL, 2017-10-25:
User zvpunry commented via [Youtube](https://www.youtube.com/watch?v=NvKreB6w_H8&lc=z22ixblzxmn4hr5ad04t1aokgr3hleaaz0juyzyni1hcrk0h00410):
> Qt5 dependency hell, but if it is properly installed it works perfectly and is easy to use
>
>One advise: Get rid of the hardware-specific code (airspy,hackrf,rtl_sdr,rtl_tcp) and focus on SoapySDR and improve on it by making it configurable from the user-interface. Anybody who wants to work with SDR hardware should use something like gr-osmosdr or SoapySDR. It is an abstraction for all SDR devices and you never need to worry about new hardware anymore. It just works. :)
>
>If you want to use something like an USRP you have to set the right Antenna-Port (RX1, RX2 or TX/RX depending on the daughterboard/USRP). Unfortunately there is some wrong port hard-coded in src/input/CSoapySdr.cpp in the CSoapySdr::restart() method. I had to change it. I searched for setAntenna and changed it from LNAW to TX/RX. I also added a call to m_device->setClockSource("gpsdo") just because i could ;)﻿

zvpunry, 2017-10-25:
Just for documentation, this is what I did: https://github.com/zvpunry/welle.io/commit/e5e04ea39445841833d8d324190c22d89ac0bd6d

It is just a hack to get it running with an usrp.

vsonnier, 2017-10-30:
+1. [CubicSDR](https://github.com/cjcliffe/CubicSDR) use it successfully, and the source code is very readable if you want API examples: [src/sdr](https://github.com/cjcliffe/CubicSDR/tree/master/src/sdr) is where SoapySDR usage lay, the rest is CubicSDR-specific code. The entry point would be `SoapySDRThread.cpp` on how to setup a stream, and stream IQ. `SDRDevice* / SDREnumerator*` is dedicated to device discovery and configuration.

zvpunry, 2017-10-30:
I removed some hard-coded defaults, they are now optional command line options. See: #153

It is now possible to use any SDR that is supported by SoapySDR just by adding some soapy arguments. Something like the following examples:

./welle-io -D soapysdr --sdr-driver-args driver=remote,remote=192.168.1.42,remote:driver=rtlsdr
./welle-io -D soapysdr --sdr-driver-args driver=uhd --sdr-antenna TX/RX --sdr-clock-source gpsdo

AlbrechtL, 2017-11-24:
Thanks for the comments and SoapySDR improvements.

I think the following:
99 % of the welle.io users* are using the rtl-sdr. For those it doesn't make sense to use SoapySDR as a HAL. For the maintenance it is easier to reduce the used libraries and the rtl-sdr part is working and tested. Also 50 % of the users are on Android and for that SoapySDR doesn't exists.

What I saw during the discussions is that SoapySDR is a great option to support SDR devices other than rtl-sdr. But those users are technically affine and have no problems to compile welle.io and use it under Linux. Nevertheless it would be great if someone can integrate SoapySDR into the AppImage [.travis.yml](https://github.com/AlbrechtL/welle.io/blob/3ef44b480a81df2fe1416c302ab8942537e9dffc/.travis.yml) or the Windows installer [.appveyor.yml
](https://github.com/AlbrechtL/welle.io/blob/3ef44b480a81df2fe1416c302ab8942537e9dffc/.appveyor.yml)

Another point is that someone has to maintain the SoapySDR code and give support to the different devices. I personally don't want to do this as well because the regular users are keeping me busy.

.* welle.io has >10.000 users. But only a few requested HackRF, airplay, CubicSDR, LimeSDR etc.

andimik, 2017-11-24:
10000? 

Wow

fvwcldgm, 2018-01-31:
That "over 10000" are probably for sure not the real users. Just people who downloaded the android app over google play.

https://play.google.com/store/apps/details?id=io.welle.welle

There are tons of bots that just collect apk files. Also probably about 90% of the downloads are "downlod, see error that a dongle is required, remove again". The most users dont read the description.
But also all windows, linux and mac users are not registered when downloading the software from other sources.

When there are real 10000 users on google play, the comment section looks completely different in the google play store then it does now.

zvpunry, 2018-01-31:
> That "over 10000" are probably for sure not the real users. Just people who downloaded the android app over google play.

Interesting idea, what counts as a real user. I have installed welle.io on the phone but only to try it. It is just too clumsy to use with the USB-OTG adapter, the rtl-sdr and the coax to the antenna all on the phone - but it works! ;)

I use it on a normal debian machine with the user-interface via x11vnc. I'm not really an android-user, so it is at least "over 10000 minus one". I still like the simplicity of the user-interface of welle.io, just a list of stations to select from and than it plays something. :-)

AlbrechtL, 2018-01-31:
You want numbers I give you numbers.

Till now Android Play store has 19.220 downloads and 5219 active installations.
The release page has 18.039 downloads in total.

Now you can speculate how many users welle.io really has.

BTW: You can check the release page download count yourself on [this side](http://www.somsubhra.com/github-release-stats/?username=albrechtl&repository=welle.io).

zvpunry, 2018-02-02:
I find usage statistics from "app store" or "play store" not particular trustworthy. Those stores make it extremely easy to just install a bunch of apps and then forget about it. I also wonder what counts as active installation. Does this mean that the program will be used regularly or does this mean it hasn't been uninstalled yet?

As I mentioned above, welle.io works on my phone (Nexus 5x) but it isn't much fun to use (for me) because of all the cables and stuff hanging on the phone. Instead I just use it on a normal Debian machine. I don't know what others do, I just can't imagine them walking around with a phone or tablet with a rtl-sdr and antenna attached to it. This is something you do once to see if it works. And if you thought welle.io is some internetradio app (remember: people don't read descriptions, they just look at nice looking icons that show a radio receiver) you will just wonder what a rtl-sdr is and why it doesn't work.

I like the simple user-interface of welle.io. I really like that it supports different SDRs via SoapySDR. I like  that it "just works" and I think that it would be sad if you try to care more about possibly non-existent phone/tablet users then real users that give detailed feedback via github. But maybe I'm completely wrong and there are really thousands of real users on android. As you said, it is just speculation how many users there are.

Now with that new splash screen you could also ask the users directly. Add a button labeled "The welle.io Developer needs your help" to it that opens a website with a survey and ask about the operating system they use, what sdr they use and what they like/dislike about welle.io.

Just some ideas and thoughts :)

gvanem, 2018-02-16:
I'm also a big fan of SoapySDR, but currently the `--sdrsoapy-driver-args` is an option and not a requirement. If one launches `well-io.exe --device soapysdr` (with no other options), `welle-io.exe` could do an `assert()` in `CSoapySdr_Thread::run()` if it find only a `null device`:
`SoapySDR driver=null hardware=null`.

IMHO it is rather ugly to have `assert()` in the code like this. And with `-DNDEBUG` the situation is even worse.

AlbrechtL, 2018-02-16:
@gvanem You have great ideas! Your are welcome send pull requests.

AlbrechtL, 2018-03-03:
I close this issue because there no more comments. But this doesn't mean that the discussion ends.

closed 2018-03-03 by AlbrechtL