AlbrechtL/welle.io #152: Linux (23 stations found) working, Android (10 stations found) not working

AlmightyFrog, 2017-10-30:
Android (KIW-L21, 7.1.2): 1.0-beta (36c288a)
Linux (Arch Linux, 4.13.7): 1.0-beta1 (fea3a21)
Interface: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

Using same RTL2838 Interface on both systems with exact same location of antenna.
Linux finds 23 stations, all of them are working. Android found only 10 stations, but can't play any of them.
Search is much slower on Android in comparison to Linux.

How can I debug this issue? What could be the cause?

Expected behavior: Same stations are found and able to play.

Note: Besides, to get it run under arch linux I did need to install manually qt5-quickcontrols2

AlbrechtL, 2017-10-30:
Which Android device did you use?

AlmightyFrog, 2017-10-30:
I have a KIW-L21 (Huawei Honor 5X) with current LineageOS running.

AlmightyFrog, 2017-10-30:
Ok, interesting point: I just now moved into next city and did test there again. Now on both devices are more stations found (Android 35 vs Linux 40) and most of the stations on Android are able to play.

I didn't change anything at the setup, just moved it. Could it be, that sensitive level or power or whatever is on laptop stronger/better than on Android?

AlbrechtL, 2017-10-31:
The Honor 5X is powerful enough.
Which RTL-SDR with which tuner (e.g. R820) do you are using?

AlmightyFrog, 2017-11-04:
Good question, how can i find that out without opening the stick?

AlbrechtL, 2017-11-05:
Just post the command line output under Linux. It is written there.

AlmightyFrog, 2017-11-05:
Thanks for pointing it out to me, here we go: "Found Rafael Micro R820T tuner"

AlbrechtL, 2017-11-05:
The R820T is the best tested tuner. Everything should be OK with your setup. It can be that your smart phone disturbs the signal (e.g. noise on the USB port).

AlbrechtL, 2017-11-24:
Closed because no more information or questions.

closed 2017-11-24 by AlbrechtL