AlbrechtL/welle.io #157: Mention rtl-sdr driver version for Android

andimik, 2017-11-08:
Seems welle.io under Android needs a newer version of rtl-sdr-driver than wavesink. I had 2.2 installed and the device did not start.

Then I found a new version even on https://f-droid.org/packages/marto.rtl_tcp_andro/ and this works now.

So please mention that version 3.06 is necessary and also that users can download it from https://f-droid.org/repo/marto.rtl_tcp_andro_20.apk or https://github.com/martinmarinov/rtl_tcp_andro-/blob/master/app/app-release.apk instead of Playstore (for those who don't have a Google account, like myself).

AlbrechtL, 2017-11-09:
Thanks for the clarification. Yes, the rtl-sdr driver got some improvements for the upcoming welle.io version.

I don't think that the rtl-sdr driver version is a big deal because the most users are using the Play store and new users will install the latest rtl-sdr driver version. Only users who already installed the old driver version and not using the Play store are affected.

I you like you can update the [README.md](https://github.com/AlbrechtL/welle.io/blob/bac547c04a855c60f533e6abb64d4822ca4328cf/README.md#download) download section with the driver information.

andimik, 2017-11-12:
done

see #159

closed 2017-11-12 by andimik