AlbrechtL/welle.io #165: [Android] Background service doesn't stop on GUI exit

AlbrechtL, 2017-11-27:
When the user exits welle.io the main process is exits but the service is still running. This creates notifications once in a while (most likely after a wake up).

[image]

Looking to the process shows the the service is not exited.

   ```
*** welle.io runs ***
libra:/ $ ps | grep welle                                                                                                                     
u0_a107   1030  415   1141192 104708 SyS_epoll_ 0000000000 S io.welle.welle
u0_a107   30678 415   1032888 34428 SyS_epoll_ 0000000000 S io.welle.welle:qt

*** After welle.io exit ***
libra:/ $ ps | grep welle                                                                                                                     
u0_a107   30678 415   1032888 34148 SyS_epoll_ 0000000000 S io.welle.welle:qt
   ```

AlbrechtL, 2018-01-08:
@twoofseven You wrote me that the services stopping is not working. Are the bugs [QTBUG-58471](https://bugreports.qt.io/browse/QTBUG-58471) and [QTBUG-62644](https://bugreports.qt.io/browse/QTBUG-62644) related with this issue?

twoofseven, 2018-01-10:
That looks promising. Especially QT UG-58471 seem to be exactly what I was dealing with. And the comments in QTBUG-62644 are also useful.

But I will not be able to verify them, before mid of next week.

twoofseven, 2018-01-15:
Update:
I found time to apply all hints within QTBUG-58471 & QTBUG-62644 and compiled welle.io with Qt v5.9.3. Unfortunately the welle.io process is still running after stopping the service. I also separated the app in different processes, but this wasn't successful.
And with Qt v5.10 the app does not even start on Android.

AlbrechtL, 2018-01-15:
Thanks for trying. I suggest that you open a new QT bug report to get help from the QT community.

twoofseven, 2018-01-20:
@AlbrechtL I found a workaround using "System.exit(0);" within the DabService to ensure that the process is terminated as well.

AlbrechtL, 2018-01-21:
A huge thanks! Soon I will publish 1.0rc2 and update the Play store version.

closed 2018-01-21 by AlbrechtL