AlbrechtL/welle.io #167: 20171007_Android_armv7_welle-io.apk package doesn't compatible with Kitkat 4.4.4

Tibor69, 2017-11-28:
I'm using the official version: 1.0-beta.
I tried installation of the 20170917_Android_armv7_welle-io.apk and 20171007_Android_armv7_welle-io.apk on my mobile and car head unit, but I got this message while installation:
Parse error / There was a problem while parsing the package.
My head unit details:
RK3066 chip Dualcore 1,6GHz CPU. 800x480 dot display, Kitkat 4.4.4 rooted OS.

AlbrechtL, 2017-11-28:
Unfortunately, we skipped the Android 4.x support. The reason is because we need newer APIs to use welle.io in the background.

I hope you can update your head unit to >= Android 5.0

closed 2017-11-28 by AlbrechtL

Tibor69, 2017-11-29:
This is wrong newsfor me, because my hardware of the headunit doesn't support the android 5.0, only support 4.x.x.

AlbrechtL, 2018-03-08:
Just to let you know. We introduced the Android 4.3 and 4.4 support again in [version 1.0](https://github.com/AlbrechtL/welle.io/releases/tag/V1.0).