AlbrechtL/welle.io #170: welle.io 1.0rc1 on Ubuntu 14.04 LTS: libstdc++.so.6 issues - CXXABI_1.3.8 / GLIBCXX_3.4.2 / CXXABI_1.3.9 not found

ghmon, 2017-12-14:
Suggestion: statically link libstdc++ and its dependancies?

Thanks - program looks great (from the screenshots)!

-- ghmon

AlbrechtL, 2017-12-14:
Did you try the AppImage?

ghmon, 2017-12-14:
Yes, messages occurred when trying to start the appImage.

me@me-PC:~/Downloads$ welle.io_1.0-rc1_Linux_x86_64.AppImage 
welle.io_1.0-rc1_Linux_x86_64.AppImage: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by welle.io_1.0-rc1_Linux_x86_64.AppImage)
welle.io_1.0-rc1_Linux_x86_64.AppImage: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by welle.io_1.0-rc1_Linux_x86_64.AppImage)
welle.io_1.0-rc1_Linux_x86_64.AppImage: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by welle.io_1.0-rc1_Linux_x86_64.AppImage)

AlbrechtL, 2017-12-14:
Do you have a 32-bit or 64-bit Ubuntu installed?

ghmon, 2017-12-14:
64 Bit (x86_64).

AlbrechtL, 2017-12-15:
@probonopd Can you see why the AppImage is not working?

probonopd, 2017-12-15:
https://github.com/AlbrechtL/welle.io/blob/98116741807ec02055448026237308b21453c4df/.travis.yml#L8

is pulling in a too new `libstdc++.so.6`.

Can you do with the compiler that comes with Ubuntu 14.04? If not, check

https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#libstdcso6

AlbrechtL, 2017-12-15:
We need C++14 features and the Ubuntu 14.04 GCC is to old to compile welle.io.

@ghmon Can you update to Ubuntu >= 16.04?

probonopd, 2017-12-15:
Updating Ubuntu would make it worse. Please don't.

ghmon, 2017-12-15:
> Can you update to Ubuntu >= 16.04?

No update at the moment (LTS) - will update to 18.04 LTS first dot release, so not before probably Jul/Aug 2018. I could boot a live CD... several here. I see the problem you have... Thanks for the help. Couldn't you also include the libs the newer libstdc++ needs in the appimage? Isn't having all libs needed bundled together is one of the advantages you have when using an appimage?

probonopd, 2017-12-15:
> Couldn't you also include the libs the newer libstdc++ needs in the appimage? 

@ghmon it is described in https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages#libstdcso6.

> Isn't having all libs needed bundled together is one of the advantages you have when using an appimage?

No, we are usually not bundling essential system libraries that can be expected to be on every target system.