AlbrechtL/welle.io #171: DAB Slideshow not appears

olilesp, 2018-01-04:
Hi !
I use welle.io for days and i see no slideshows appears into the software. 
My system is W10  and latest version of welle.io
 RTL SDR 2832u r820t2

andimik, 2018-01-04:
Please note that slideshows are used in every DAB ensemble. 

You didn't mention where you are living and what ensembles you mean. For details please see http://www.fmlist.org/sendertabelle/dab-ww.php

olilesp, 2018-01-04:
Oh, sorry, i miss these things : 
I live in belgium, receiving RTBF DAB 12B.
RTBF DAB provides some slideshows on Classic 21 - Musiq3 - Pure and Vivacité, but i can't receive them.

andimik, 2018-01-04:
Ok, then I know why. See answer from Andi Gsinn 

http://radioforum.foren.mysnip.de/read.php?35326,1171841,page=129

Belgium uses separate data services. 

Maybe you can record a rawfile or even an eti-file and post the link.

olilesp, 2018-01-04:
OK.
Here's a dump (2 files).
https://we.tl/hXCCESb7GF

And if you wish a french translator, i'll keep this job ! ;) 

Thx !

AlbrechtL, 2018-01-05:
@olilesp Thanks for the RAW files. I will analyze the issue.

Regarding your French translation offer. C'est super! Please follow the description at https://github.com/AlbrechtL/welle.io/wiki/Translating-welle.io and send a pull request.

AlbrechtL, 2018-01-05:
I took a short look into the RAW file "rtbf_3.sdr". The stations Classic 21, Musiq3, Pure and Vivacité are classic DAB (not DAB+) stations. As far as I know no classic DAB station are sending MOT slide show.

Do you have a receiver that can display the slide show?

AlbrechtL, 2018-01-07:
I did a little more analyzing.

After adding the option “--mp2-file” it was possible to analyze the X-PAD data with [XPADExpert](https://www.basicmaster.de/xpadxpert). For the station “Classic 21” XPADExpert shows that there is only dynamic label included but no MOT slide show.
[image]

For that reason I close this issue.

closed 2018-01-07 by AlbrechtL

gvanem, 2018-01-07:
*BTW*.<br>
Could the ANSI-codes for red and yellow `[X-PAD len]` in `pad_decoder.cpp` be fixed to work for Windows too?

olilesp, 2018-01-09:
Hi
Sorry for the delay,
RTBF DAB sended those slidewhows regular, some receivers show them (sangean radios in exemple).
But...  In Belgium (VRT DAB, Norkring DAB, RTBF DAB and the tests networks on Liege), the slideshow is sended in a different data stream (not in PAD) but not detected by welle.io. You can only see the Slideshow when you select the SLS Datendienst (Vivacite / Classic 21 / Pure in ex.) .

AlbrechtL, 2018-01-10:
@basicmaster Do you know how the extra data stream is signaled inside the station stream?

@olilesp Please open a new pull request for your French translation.

@gvanem Please open a new issue because your problem is not related with this issue

basicmaster, 2018-01-10:
Signalling for this separate data stream is done in the FIC:
- FIG 0/2 signals an additional packet data component for the service, with a ScId
- FIG 0/13 signals that this component contains a Slideshow (actually this was also required for X-PAD slideshow, but is currently not required in DABlin's code, as it is implicitely detected just by its presence in the X-PAD, to keep it simple)
- FIG 0/3 maps the ScID to a specific subchannel and a specific packet address (it is possible to have more than one transmission on the same packet data subchannel)

All this has to be combined and furthermore the packet mode has to be implemented, to extract the respective data from the packet mode subchannel. So this is a bit complex ;-)

So far I was not aware of any DAB services that currently use a separate subchannel for the Slideshow (in Germany this was done some years ago for 90elf). Therefore I did not yet add support to DABlin's data service processing. But it seems that there is demand for it ;-)

It would be helpful to have a longer sample, e.g. 4 minutes, to test this with more than one slide.

AlbrechtL, 2018-01-12:
Thanks for the explanation. I added this task to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

@olilesp It would be nice if you could provide a 5 minutes RAW file sample.

olilesp, 2018-01-13:
Here's the Raw... IQ Format 1,5Go 
[https://we.tl/1JFOKw6VmP](url)

AlbrechtL, 2018-01-13:
I added a special Belgium French handling in edc258d. Feel free to test it.

AlbrechtL, 2018-02-14:
andimik pointed out that Beligum will switch to DAB+ this year (https://www.worlddab.org/country-information/belgium). Thus I think there is no need to implement such special MOT handling.

olilesp, 2018-02-15:
RTBF has annonced on the "radio day" that DAB+ tests and slideshows are on Test Classic 21+ , i've sended a mail to RTBF to ask when they would propose MOT slideshows and data like other countries and they reply that configuration will continue... VRT DAB + and DAB VL also keep this configuration.

basicmaster, 2018-02-15:
Could you ask them, for what reason they have the slideshow in a separate subchannel? To me this doesn't really make sense (unless they used it for more than one audio service).

andimik, 2018-02-15:
Remark: I have samples from old DAB from Südtirol at home. There is was in the same subchannel...

olilesp, 2018-02-23:
Yes, Belgium is the only country where another method to send MOT SLS is used... There's two normalisations for SLS and RTBF - VRT - Norkrign use the 2d. And this will continue.
Norkring and VRT are regular transmissions, RTBF is again experimental (from 1997 !) and they're testing again DAB+
Classic 21 (DAB) and Test Classic 21 + (DAB+) use the MOT SLS and they won(t change anything because this is working with portable radios. I remember that in 2000 and more DAB at RTBF used PAD in the same channels to display SLS and MOT for a website.  They also know that's a specific DAB usage... no change is allowed at this time.

Some dab players in dev are updated th show SLS but there's a long work to implement this.
Hope that welle.io that i'm using on PC and testing on my Android in few days will show them in the near future :)

I've another question about sensibility reception, i'll open a subject into the forum next days. 

Thanks for your work and sorry if i can't speak english very well.