AlbrechtL/welle.io #176: 56 kbps MP2 not playing

es5nhc, 2018-01-24:
I am reporting trouble with playback with a low bitrate DAB station in Devon multiplex in the United Kingdom. Here's a raw file which can be used to reproduce the issue: https://www.dropbox.com/s/7i1pj8grlox3fi7/SDRTouch_20180124_022829_220352kHz_IQ.wav?dl=0

While other stations play fine, the station called EXE (as in Radio Exe) will only play occasional glitches and is mostly silent. at the same time, the software says into command line:
`56 (3) has a problem`

Other stations on the multiplex are fine. And yes, the station in question really is transmitting 56 kbps MP2 mono. Ugh!

The software version I am using is welle.io 1.0-beta

andimik, 2018-01-24:
Well, this surely comes from Jan's code. I have no chance in welle.io, qt-dab, dab-cmdline and eti-stuff.

If I play it in qirx under Windows (see http://softsyst.com/QIRX/qirx ) , then I can listen to "You can get it if you really want it".

In case you work under Linux could you record an eti-file using dab2eti? See https://github.com/linuxstb/dabtools . Only then we could investigate more. 

But thanks for the rawfile anyway, I'll double checked it with DABLIST.org regarding the TII code 0x3c 1409, and it seems this is transmitter Exeter/Saint Thomas. But the content is not up-to-date. Is EXE a new station there?

AlbrechtL, 2018-01-24:
I confirm this bug. Thanks for reporting!

andimik, 2018-01-24:
I fear the problem is caused by the unusual sampling rate of 24 kHz.

Because with qirx I got a wav-file having this SR.

AlbrechtL, 2018-01-24:
Fixed in 1aca727. The 56 kbit/s audio service component protection profiles was missing.

@JvanKatwijk You may can fix this issues in your software as well.

closed 2018-01-24 by AlbrechtL

es5nhc, 2018-01-24:
@andimik , Yes, EXE was added some time last year.

Here's an ETI: https://www.dropbox.com/s/qm7o8cdr8o4rxcz/devon.eti?dl=0

A clue could have been that welle.io thinks that EXE was stereo while it actually was mono. 

Thanks for the quick fix.