AlbrechtL/welle.io #177: Support for multiple SDR-sticks

gvanem, 2018-01-28:
I got a new and improved [RTL-SDR stick](https://www.amazon.com/gp/product/B011HVUEME). Now having 2 sticks connected to my PC, I see no way in Welle-IO to select a specific one; it always iterates over `rtlsdr_open(&device, i)` and selects the first found.
I made a feeble attempt to fix it (currently only for the `CRTL_SDR` class). Patch:
https://gist.github.com/gvanem/8ebc16da2eb367aec3f0ef5f2544c675

I added an `-i` (instance) option to `welle-io.exe`, which now prints:
```
  -i <Instance>           Instance number for airspy, rtl_tcp, rtl_sdr if you
                          have more than one of each (default: 0)
```
Would be great if this could be supported.

*PS*. since I moved my old RTL-SDR stick to another USB-port some weeks ago, I have this situation:
```
c:\>rtl_tcp
Found 3 device(s):
  0:  , H☻, SN:
  1:  Realtek, RTL2838UHIDIR, SN: 00000001
  2:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
usb_open error -12
Failed to open rtlsdr device #0.
```
Starting `welle-io.exe`  with no arguments will correctly select `device 1` (not the *junk* device 0 as some other `rtl_*.exe` programs would). Since my new stick has the same RTL2832U interface (and Rafael Micro R820T tuner), the *zadig* driver does not need a reinstall. Is that correct? But how do get rid of this junk-device `0`?

AlbrechtL, 2018-01-29:
Thanks for the improvement!
What is the use case to have two dongles connected with the PC?

Duplicated with #95.

gvanem, 2018-01-29:
> What is the use case to have two dongles connected with the PC?

Both have different frequency coverage and antennas for different bands (but some overlaps). Thus without physically unplugging one of them, it would be useful to support several instances. 

Maybe a `--device rtl_sdr[:N]` syntax would be better?

AlbrechtL, 2018-01-29:
I would prefer something like `--device-index`. Then each driver can implement this options if needed.
Please send a [pull request](https://help.github.com/articles/creating-a-pull-request/) with your changes.

AlbrechtL, 2018-03-03:
I added this topic to [Open Tasks](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

closed 2018-03-03 by AlbrechtL