AlbrechtL/welle.io #178: Enable automatic dvb_usb_rtl28xxu detaching on linux builds

fvwcldgm, 2018-01-29:
I download welle.io from here: https://github.com/AlbrechtL/welle.io/releases
I got the AppImage file. Made it with chmod a+d *.AppImage executable and ran it. 

The software tells me that it cant find any device.

The logfile tells me:
Kernel driver is active, or device is claimed by second instance of librtlsdr.
In the first case, please either detach or blacklist the kernel module
(dvb_usb_rtl28xxu), or enable automatic detaching at compile time.

The default setting of gqrx for example automaticly detaches the driver, run the software and attaches it at closing again - and gqrx is much more complicated software by default for "normal user".

Why is welle.io that provides AppImage files to make everything as simple as possible requesting from user to read logfile-outputs and detach kernel drivers by hand?
The default setting should be "automatic detaching as default".

AlbrechtL, 2018-01-29:
Which Linux do you use?

In case of a Debian based system (e.g. Ubuntu) just install the package `librtlsdr0`.

fvwcldgm, 2018-01-31:
Its installed already as dependency from gqrx.
When you install on for example a Debian based system with "apt-get install gqrx" the gqrx software, you have afterwards librtlsdr0 (Version 0.5.3-13) installed. gqrx is as written in the first post able to unload and load the driver automatically. On the same system on the same day after install gqrx i downloaded welle.io_1.0-rc2_Linux_x86_64.AppImage and ran it. It told me in the logfile that automatic detaching should be compiled into the image (you see the logfile above).

Normal users should not have to be told to compile their own software versions. Gqrx is much more complicated software but dont tell their user to compile automatic detaching of modules by their own.

Please set a flag that is been read out at runtime that is been set at compile time. The default should be "automatic detaching = on". If this flag is turned to "automatic detaching = off" at compile time but dvb_usb_rtl28xxu is running the software should just report at no working device detection "ERROR: dvb_usb_rtl28xxu module is running but the software is been compiled with "automatic detaching = off". Please unload dvb_usb_rtl28xxu by hand."

This would also fix the problem of not knowing the real failure reason we have now: Quote: "Kernel driver is active, or device is claimed by second instance of librtlsdr."
Seeing an active dvb_usb_rtl28xxu module but "automatic detaching = off" would be then something easy to understand from the logfiles and not an "or" situation we have now.

AlbrechtL, 2018-01-31:
Which Linux do you use?

fvwcldgm, 2018-02-03:
I use many different GNU-Linux OS.
For the test Debian have been used.

mpbraendli, 2018-04-01:
Short investigation: This seems to be done in `librtlsdr`, see https://github.com/steve-m/librtlsdr/blob/master/src/librtlsdr.c#L1502 and it depends on how the library was compiled.
Debian seems to compile with DETACH_KERNEL_DRIVER=ON, see https://sources.debian.org/src/rtl-sdr/0.5.3-13/debian/rules/

However, is librtlsdr included in the AppImage? If so, is it also built with DETACH_KERNEL_DRIVER=ON ?

mpbraendli, 2018-08-28:
No progress. Please open issue again if still a problem.

mpbraendli, 2018-09-02:
Is still an issue, seen today during EMFcamp workshop on several machines, debian and ubuntu.

AlbrechtL, 2018-09-13:
Today I remembered that @probonopd mentioned this issue in https://github.com/AlbrechtL/welle.io/issues/45 and Jan (qt-dab) and he discussed a possible solution in https://github.com/JvanKatwijk/qt-dab/issues/34.