AlbrechtL/welle.io #179: Increasing and Decreasing the gain more then 100 times in a minute

fvwcldgm, 2018-01-29:
When i have low quality signal, i get more then 100 times in a minute this in the logfile. I dont think that it makes no sense to jump so extreme fast between two gain values.

RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
[X-PAD len] [X-PAD len] [X-PAD len] RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
[X-PAD len] [X-PAD len] [X-PAD len] RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
[X-PAD len] RTL_SDR: Decreased gain to 37.2
[X-PAD len] [X-PAD len] RTL_SDR: Increased gain to 38.6
[X-PAD len] RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2

AlbrechtL, 2018-01-29:
Duplicated with #74.

closed 2018-01-29 by AlbrechtL

fvwcldgm, 2018-01-31:
Why have you closed this when the other bug that seems to still exist is also closed? I have seen that you entered into the #74 that you have added this to a todo list. Normaly bugs are closed when they are done. Not when they are on a todo-list but not done now.
Most people dont look at duplicates on opened bugs. I have done the bug-reporting correctly and have looked through the opened bugs and was not able to find this so i reported it.
You cant expect that the users look through all of the closed bugs and in this case then reopen those - when its known that they are not fixed now.

AlbrechtL, 2018-01-31:
I closed it because it is not a bug. The behavior is not really nice but it is intended because a hysteresis implementation is missing. This is the reason why it is on the to do list.

fvwcldgm, 2018-02-03:
Cant you just add the increasing and decreasing functionality on this logic?:
If "perfect" gain seems to be 37.9 but we are about - or + 2(or more) from this value away, then jump to the better one.
Then we wont have this 
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
RTL_SDR: Increased gain to 38.6
RTL_SDR: Decreased gain to 37.2
the whole time because the logic "beeing 2 or more away" wont be true any more. Maybe also - or + 1.5 would be even enough to stop this jumping around.

AlbrechtL, 2018-02-04:
Honestly, I do this project in my spare time. Even small changes needs time to do it. This is a low priority change. So if you are a developer you are free to fix it and send a pull request.