AlbrechtL/welle.io #183: Android Welle.io starts randomly

Dtrieb, 2018-01-29:
After the last Update (beta) from the PlayStore welle.io starts randomly without anything. I try to kill the process but welle.io starts sometimes again.

Dtrieb, 2018-01-29:
Same like here?
https://github.com/AlbrechtL/welle.io/issues/165

AlbrechtL, 2018-01-29:
@twoofseven Can you check what is going wrong?

Dtrieb, 2018-02-03:
I was closing welle.io with the exit button.
It reopen randomly with not usb stick (DVB-T, or Power) 

How can I make you a adb log, if you think it would help?

fvwcldgm, 2018-02-03:
May this helps you to get able to provide logfiles:
https://developer.android.com/studio/command-line/logcat.html

twoofseven, 2018-02-03:
You can also use the bug report function within the developer settings: https://youtu.be/YxQ3NTloD2c

pr2git, 2018-02-21:
Just to confirm this bug and I think that it is linked with the network connection. If Android device lost the network connection when the network is back it seems to me that well.io starts automatically by itself. So when network is back there is a trigger that restart well.io.
Question is also why well.io is running in the background? Why not stop it when we exit?

Dtrieb, 2018-02-28:
I didn't find anything in the log file. But I see that it often starts again if I run the play store.

andimik, 2018-12-11:
Is there maybe an update checker?

AlbrechtL, 2018-12-12:
No, there is no checker.

But in welle.io 2.0 this problem will not be occurring because I removed the code for background playing. This issue was one reason to mark the background playing code as non-maintained.