AlbrechtL/welle.io #184: [RPi] Mouse cursor always present when using touch screen

FartherOut, 2018-01-31:
Minor visual bug/annoyance again, but is there a way to hide the cursor when using a touch screen?

fvwcldgm, 2018-01-31:
The software should not remove the mouse cursor by default. It should be more likely removing the mouse after some time when run on full screen and displaying it again when the mouse is beeing moved.
Just like its done on for example YouTube fullscreen mode.

For additional style-points there could be build in the detection of the input device.
For example after few Seconds the mouse is not visable any more like on YouTube. When you use a touchscreen as input, it should not be displayer any more. If you move your for example PS2 or USB mouse, then it should be displayed as long as the movement is running and should dimm away after few  moments when stopped moving.

This configuration can be then used on all Systems - with or without touchscreen and would fullfill all the design requirements on both systems.

FartherOut, 2018-01-31:
This sounds very nice, and would be handy to have.

AlbrechtL, 2018-03-03:
Funny in #201 an users complains the opposite under Windows.

AlbrechtL, 2018-04-07:
From our point this has to be handled by the graphic system.

closed 2018-04-07 by AlbrechtL