AlbrechtL/welle.io #186: Retrieving data from a frame.

MaxymValeriyovych, 2018-02-02:
Hello. Help is needed. Where the data after decoding is saved in the code (Mscblock)? Where the data for playback by the sound card is saved in the code?

AlbrechtL, 2018-02-02:
What are you trying to do?

MaxymValeriyovych, 2018-02-02:
I want to get data from buffer. But i don't search when this place in code

AlbrechtL, 2018-02-04:
The best option would be to run welle.io inside a debugger and find the data that you are looking for.

closed 2018-02-04 by AlbrechtL

MaxymValeriyovych, 2018-02-05:
void	ofdmDecoder::decodeMscblock (DSPCOMPLEX *vi, int32_t blkno) {
	bufferSpace. acquire (1);
	memcpy (command [blkno], &vi [T_g], sizeof (DSPCOMPLEX) * T_u);
	helper. lock ();
	amount ++;
	commandHandler. wakeOne ();
	helper. unlock ();
}

I can not find in code, where is a buffer with data bytes of mscblock