AlbrechtL/welle.io #187: Android Welle.io using rtl_tcp

Pooyankh, 2018-02-02:
I just learned that Welle.io appears on the list of applications that my car can use (Mirrorlink via Rockskout).

I can control the welle.io, but the problem is I cannot plug in the RTL-SDR directly to my Android phone, as the USB port is already connected to my car screen in client mode.

There is no option to use rtl_tcp, so I can setup a raspberry pi and remote rtl senario.

Any comments?

AlbrechtL, 2018-02-04:
The Android version uses the rtl_tcp driver internally. If you are able to compile welle.io you can try to change the IP-adress [here](https://github.com/AlbrechtL/welle.io/blob/41ee4ba228b018a104d2ab63c3113ccecd2bf5b8/src/CRadioController.cpp#L193).

Pooyankh, 2018-02-05:
Thanks for info. Any chance the IP / port can be exposed as a user editable setting under the advanced settings menu?

AlbrechtL, 2018-02-05:
I the moment this functionality is not available. But I added it to the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

closed 2018-02-05 by AlbrechtL

lars18th, 2018-02-26:
Hi @AlbrechtL ,

I hope to see this soon. I like to connect my Android device to my remote RTL_TCP server.
Any option to support a "config.file" meanwhile?

Thank you for this great DAB Player!
Regards.