AlbrechtL/welle.io #189: Problems with limesdr/soapysdr

freqmod, 2018-02-03:
I had some problems using welle.io with limesdr:

1. Set frequency causes limesdr to run out of usb transfer buffers.
Furst time welle.io tunes i get a signal, next time i only get errors. I have worked around this by reinitializing soapysdr every time setFrequency is called:

void CSoapySdr::setFrequency(int32_t Frequency)
{
    m_freq = Frequency;
    if (m_device != nullptr) {
        stop();
        restart();
        setFrequency2(Frequency);
    }
}

void CSoapySdr::setFrequency2(int32_t Frequency)
 {
     m_freq = Frequency;
     if (m_device != nullptr) {
         m_device->setFrequency(SOAPY_SDR_RX, 0, Frequency);
        /*Frequency = m_device->getFrequency(SOAPY_SDR_RX, 0);
        qInfo() << "OutputSoapySDR:Actual frequency: " <<
             Frequency / 1000.0 <<
             " kHz.";
            */
     }
 }
2. 
Auto gain for limesdr doesn't work. Setting manual gain works, but it is reset every time soapy is reinitialized (on every frequency switch if the fix above is implemented).

------
Info about system:

$ SoapySDRUtil --info
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Lib Version: v0.6.1-g4daa4329
API Version: v0.6.0
ABI Version: v0.6

----

[INFO] Make connection: 'LimeSDR-USB [USB 3.0] xxxxxxxxxxxxxx'
[INFO] Reference clock 30,72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 3,072e+07 MHz
[INFO] LMS7002M calibration values caching Disable
SoapySDR driver=FX3 hardware=LimeSDR-USB  boardSerialNumber=0xxxxxxxxxxxx firmwareVersion=4  gatewareVersion=2.12  hardwareVersion=4  protocolVersion=1
SoapySDR master clock rate set to  32768  kHz
OutputSoapySDR:Actual RX rate:  2048  ksps.

mpbraendli, 2018-02-04:
@AlbrechtL I can have a look at this one as soon as I'm back home and have my LimeSDR with me. You can assign the ticket to me.

freqmod, 2018-02-04:
Seems like the frequency tuning is a problem with the newest version of Limesuite. Version commit caa459a47a194fbb6a99e6d303cf9c56d0cf3983 (HEAD, tag: v17.09.1) is good (at least it seems so from using GQRX that is also using SoapySDR).

AlbrechtL, 2018-03-03:
Is this issue still existing?

mpbraendli, 2018-03-16:
I can't get it to work with the LimeSDR and with LimeSuite v18.02.0-g64c6155e
There is a very strong DC offset, and the AGC pushes the signal to very low amplitude.

mpbraendli, 2018-04-01:
Fixed command line parsing, manual gain settings and enabled DC offset compensation, now I can receive again with my LimeSDR.
Commit 7082c28 in branch `next`.
Next, I will try to get the Software AGC to work.

closed 2018-04-01 by mpbraendli