AlbrechtL/welle.io #190: Jerky and out of tune audio

freqmod, 2018-02-03:
Several stations in the following clip experiences strange audio (e.g. Radio nova in Oslo, Norway). 
Recorded with:
rx_sdr -d driver=lime,antenna=LNAW -g LNA=20,PGA=20 /tmp/t -f 232496000 -s 2048000 
and compressed with xz. 
Decompress with xz and then listen in welle with:
./welle-io -d rawfile --raw-file <filename>

https://drive.google.com/file/d/1SonXfIH2l1nWW7wKrsiBZJ0Ud2ayWQSx/view

AlbrechtL, 2018-02-03:
Thanks for reporting and the RAW file.
Approved this is a bug.

andimik, 2018-02-03:
Radio Nova uses the worst protection level 4B. It has lots of frame errors. Please first try to get a better signal.

andimik, 2018-02-03:
So, this page is outdated: http://www.fmlist.org/sendertabelle/dab-ww_more.php?inputLand=NOR&inputEn=FA61

In the meanwhile they changed the Mux, some stations got the best PL 1A, but some only the worst (4A and 4B).

andimik, 2018-02-03:
[norwegen_lokalradio.txt](https://github.com/AlbrechtL/welle.io/files/1692110/norwegen_lokalradio.txt)

AlbrechtL, 2018-02-03:
The strange thing is that the DAB ensemble signals the data rate e.g. "Radio Nova" at 64 kbit/s (FIG0 extension 1) but the AAC codec signals it at 32 kbit/s. This explains the to fast audio output.

andimik, 2018-02-04:
I'm using f9b7164 (beta1). Sound has some dropouts, but the speed is ok.

Seems later version have this bug (just tried latest nightly build)

AlbrechtL, 2018-02-04:
Can you check in which version the problem begun?

andimik, 2018-02-10:
Same problem in Vienna. I have a pre-recorded file from there.

I've checked all appimages from https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

BTW: I just realized you have changed the command line options in 2140350 ...

**Last good version** playing all stations in the correct speed was 2140350 dated 26th Jan 2018
>  ./20180126_2140350_Linux_welle-io-x86_64.AppImage -d rawfile --raw-file foo.raw 

**First bad version** is 180a945 dated 28th Jan 2018
> ./20180128_180a945_Linux_welle-io-x86_64.AppImage -d rawfile --raw-file foo.raw 

Starting with this 180a945 or later this gives lots (or tons ...) of errors like these:

> qrc:/src/gui/QML/style/TouchComboBox.qml:30: ReferenceError: Unit is not defined

or 

> file:///tmp/.mount_2018019F0jyY/usr/qml/QtQuick/Controls/SplitView.qml:541: ReferenceError: parent is not defined
> qrc:/src/gui/QML/RadioView.qml:14: TypeError: Cannot read property of null

Even `-h` produces lots of errors!

> (20180208_e996357_Linux_welle-io-x86_64.AppImage:5529): Gtk-WARNING **: Theme parsing error: gtk-widgets-assets.css:76:7: Missing name of pseudo-class
> 
> (20180208_e996357_Linux_welle-io-x86_64.AppImage:5529): Gtk-WARNING **: Theme parsing error: gtk-widgets-assets.css:80:7: Missing name of pseudo-class
> 
> (20180208_e996357_Linux_welle-io-x86_64.AppImage:5529): Gtk-WARNING **: Theme parsing error: gtk-widgets-assets.css:84:11: Missing name of pseudo-class
> 
> (20180208_e996357_Linux_welle-io-x86_64.AppImage:5529): Gtk-WARNING **: Theme parsing error: gtk-widgets-assets.css:90:11: Missing name of pseudo-class
> 
> (20180208_e996357_Linux_welle-io-x86_64.AppImage:5529): Gtk-WARNING **: Theme parsing error: gtk-widgets-assets.css:96:7: Missing name of pseudo-class

Has been only partly (!) fixed in https://github.com/AlbrechtL/welle.io/commit/368d9fb8f504de97ea815b251fe8f8b23d1fb1bf

Tons of errors still appear in e996357 (see quote above)

I guess it has to do with sampling rate other than 48 kHz together with qt-quick 2.6 👎

AlbrechtL, 2018-02-10:
Thanks! Fixed in 0e9abc90daf06983b3ff76252eb87d52ee3f7751.

closed 2018-02-10 by AlbrechtL