AlbrechtL/welle.io #193: rtl_tcp use wrong port

betaros, 2018-02-05:
I created a shortcut to use welle.io with my SDR stick connected on my Odroid-c2 over the network. 
But welle.io always uses the wrong port 1234.

The command I use in the shortcut is:
> "C:\Program Files (x86)\welle.io\welle-io.exe" -D rtl_tcp -I 192.168.178.25 - P 1337

[image]

gvanem, 2018-02-05:
Which version is this? The `-P` option was dropped some time ago. Try ` --rtltcp-port 1337` instead.

betaros, 2018-02-05:
The version I use is 1.0-rc2.

When I try
> "C:\Program Files (x86)\welle.io\welle-io.exe" -D rtl_tcp -I 192.168.178.25 --rtltcp-port 1337

I get:
> Unknown option 'rtltcp-port 1337'.

Another idea was
> "C:\Program Files (x86)\welle.io\welle-io.exe" -D rtl_tcp --rtltcp-ip 192.168.178.25 --rtltcp-port 1337

The result was
> Unknown options: rtltcp-ip, rtltcp-port.

AlbrechtL, 2018-02-05:
Please remove the white space " " between "-" and "P". It has to be "-P" not "- P".

closed 2018-02-05 by AlbrechtL