AlbrechtL/welle.io #194: Writing audio data to a file

MaxymValeriyovych, 2018-02-07:
Please, help me. How can I write audio DATA from this buffer to a file?

http://prntscr.com/ibb6ln

It's point in code.

qint64 CAudioIODevice::readData(char* data, qint64 len)
{
    qDebug() << "MyData   " << data;

    qint64 total = 0;
    total = Buffer->getDataFromBuffer(data, len / 2); // we have int16 samples
    if(total == 0)
    {
        memset(data, 0, len);
        total = len / 2;
    }
    return total * 2;
}

gvanem, 2018-02-07:
Use a `f = fopen (file_name, "wb"); ` in the constructor, a `fclose(f);`  in the destructor and a
`fwrite (data, 1, (size_t)total f);` above.
But what are you trying to accomplish?

MaxymValeriyovych, 2018-02-07:
I want to save шт file data which fed to the sound card.

MaxymValeriyovych, 2018-02-07:
I have hieroglyphs in file instead normal symbols. How to change theirs encoding?
Hieroglyphs like on screenshot http://prntscr.com/ibb6ln

AlbrechtL, 2018-02-07:
Please search at Google for a tutorial how to deal with binary data in C/C++.

closed 2018-02-07 by AlbrechtL