AlbrechtL/welle.io #196: Disable Frequency correction

lars18th, 2018-02-14:
Hi,

With a very low signal, the Automatic Frequency Correction generates a lot of troubles.
So, I suggest to add an option for disable it, or to manual set it.

Please, consider it.
Thank you!

AlbrechtL, 2018-02-14:
welle.io is an easy to use application. Such features like a manual frequency correction is really for advance users. For that I recommend [QIRX](http://softsyst.com/qirx).

closed 2018-02-14 by AlbrechtL

lars18th, 2018-02-15:
Hi @AlbrechtL ,

I suggest to re-open this issue. I'll explain the problem in more detail after using the QIRX tool:

- When I use Welle.io with my RTL-SDR dongle, it's very hard to lock the signal... because the Frequency correction it's changing continously. However, when I use the QIRX tool, I can tune more fast... and after the signal is locked, I hear all the programs well (with small glitches). The power of the signal it's very low (around 4-6db).

- However, when I record the IQ samples using the QIRX tool, and I replay the _same_ file using Welle.io in the same machine, I can play all the programs like with QIRX. The only drawback is that Welle.io takes a lot of time doing the Frequency correction... But after it "locks" the signal, it plays well. Even when the file restart.

So, comparing the behaviour of Welle.io using _direct_ RTL-SDR input and _recorded_ IQ input (from the same device, and without changing the conditions), in my opinion some trouble exists with this parameter.
What you think?

lars18th, 2018-02-15:
Hi @AlbrechtL ,

Please, discard all the comment done in my previous post. I see now the difference:

- When using Welle.io with the RTL_TCP input, instead of the internal RTL_SDR, I see best results... less errors and fast lock of the signal.

So, what I'm doing now it's use the "RTL_TCP.EXE" found in QIRX and run Welle.io with the RTL_TCP source.
I don't know why, but with this I see improved results.

I hope this help! :smile:

gvanem, 2018-02-15:
@lars18th  You may know that Welle-IO can support [SoapySDR](https://github.com/pothosware/SoapySDR/wiki) devices if built with `-DHAVE_SOAPYSDR`. It's quite easy if you're a programmer. Then you'll be able to create whatever SoapySDR-module you 
wish (even write one in Python).

Please don't pull the awful QIRX project into this project; there are enough crappy SDR-programs out there.

lars18th, 2018-02-15:
Hi @gvanem ,

Sorry if my post ofens you! It's not my idea.

I'm not a big fun of SDR-programs... My objective is a **reliable** DAB client-server configuration. And in my zone DAB signals have a very low level. So, I need to run with _erroneous bitstreams_. And from my point of view Welle.io it's one of the **most reliable** DAB players. This, and the binary releases for different platforms (PC and Android, for example) it's a must have.

However, I discovered that the internal RTL_SDR module doesn't work equal to other RTL_TCP modules. This is the only question.

AlbrechtL, 2018-02-15:
The issue is the weak frequency synchronization we know that and we will focus on it in future. This is already on the [task list](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks) so the issue will not be reopened.

lars18th, 2018-02-16:
Hi @AlbrechtL ,

> The issue is the weak frequency synchronization we know that and we will focus on it in future. This is already on the task list so the issue will not be reopened.

Thank you for explaining the cause. Now I see the "Improve DAB frequency shift detection (frequency synchronization)" in the Open Tasks list. :+1: 

However, I ask this question: As the frequency synchronization is done _inside_ the Walle.io, why the result it's different using the internal RTL_SDR input and the RTL_TCP input? I feel it has no sense!

lars18th, 2018-02-23:
Hi @AlbrechtL,

Please, reopen this issue!

Using a perfect signal from ODR-Dabmod, when Welle.io "losts" the signal (some trouble appear), then it tries to do a frequency correction. But the algorithm **has problems**. Sometimes it enters in an infinite loop of search, never returning to 0Hz (the correct frequency).

So, please review the algorithm. Also, include an option for disable it and use a fixed offset, if the user prefers it.

Thank you for this awesome DAB player! :smile:

xenoid, 2018-08-05:
I don't know if this still is an issue, but i got the same problem.
I have "OK"-reception, but once the frequency correction starts kicking in everything goes downhill.
Some tips or tricks?

AlbrechtL, 2018-08-05:
Can you try the latest version from the next branch please.

pawol, 2023-05-16:
Auto ppm mode does not work at all. My ppm should be 68 but your program calculates 151 and no station is tuned