AlbrechtL/welle.io #198: Method for remove only one channel

lars18th, 2018-02-14:
Hi,

Some times, my channels (because of an insuficient signal) "changes" the frequency when exploring other frequencies. Then, when I re-tune the real frequency the channel it's repeated.

So, I suggest a method for removing the selected channel/program... and not the entire list.

I feel this not will be complex, right?

andimik, 2018-02-14:
I guess first please fix your reception issue...

lars18th, 2018-02-14:
> I guess first please fix your reception issue...

Yes! However, I feel very unconfortable the need to delete all channels for just remove one channel. This it's very user unfriendly.

andimik, 2018-02-14:
Anyway, this is a strange behavior. Is a service to be found in both muxes?

lars18th, 2018-02-14:
> Is a service to be found in both muxes?

No, no! The "error" is this:

- Tune one MUX, and play one program.
- The signal in the mux it's low.
- Tune another frequency _without_ any signal
- Then... sometimes the Welle.io thinks that the program is in the _new_ frequency.
- And when you return to tune the _original_ frequency the program it's duplicated.

I hope you can see the problem: The frequency of the program it's incorrectly changed. This it's a bug.
However, in any case, it's a must have the option for deleting a selected program. Don't think so?

Regards!

AlbrechtL, 2018-02-14:
Agreed it's a bug.

andimik, 2018-02-14:
Please give us more details. Which Mux you are talking about? 

Is the Mux also broadcast (somewhere else) on the empty channel? 

I do not agree with you. Please rescan in such a situation. Because the software thinks you change your position.

andimik, 2018-02-14:
Ok, but why do you select an empty channel without signal?

lars18th, 2018-02-14:
> Ok, but why do you select an empty channel without signal?

Because in fact isn't a empty frquency... it's a frequency with near empty signal.
I have a very low signal in my territory!

andimik, 2018-02-14:
Then you gave us a wrong description

AlbrechtL, 2018-02-14:
I recommend to stop this discussion. I know the bug and it will be fixed in some time.

AlbrechtL, 2018-09-01:
Fixed in https://github.com/AlbrechtL/welle.io/commit/313b0d720f8a487142f525e5525da2e5006cd384.

closed 2018-09-01 by AlbrechtL