AlbrechtL/welle.io #200: Windows command line parameters error?

lars18th, 2018-02-15:
Hi,

I'm trying to use the rawfile input source with the Windows version. However, when I execute:

`welle-io.exe -d rawfile --raw-file c:\tmp\file.raw --raw-format u8`

I receive one window error with this info:

`Unknown options: d, raw-file, raw-format`

Please, can you help me?

lars18th, 2018-02-15:
I found the response...

`welle-io.exe -D rawfile -F c:\tmp\file.raw`

It's documented here: http://www.welle.io/devices/rawfile#odr-dabmod
So, I suggest to change the README.md of the github homepage, as the info in #USAGE is inconsistent.

Regards.

closed 2018-02-15 by lars18th

andimik, 2018-02-15:
Please use a newer version or replace the parameters by

-F 
-D

lars18th, 2018-02-15:
> Please use a newer version or replace the parameters by

newer than 1.0-rc2? https://github.com/AlbrechtL/welle.io/releases
I'm using this version.

AlbrechtL, 2018-02-15:
The page https://github.com/AlbrechtL/welle.io reflects the latest development stage.
The source code of 1.0rc2 is [here](https://github.com/AlbrechtL/welle.io/tree/4bf9a9256b290cea6fdb8b2ab02ceb52dc15b451) which also included valid command line options for this version.

AlbrechtL, 2018-02-15:
BTW. You can download latest build here: https://bintray.com/albrechtl/welle.io/welle.io_nightly#files

lars18th, 2018-02-16:
Thank you, @AlbrechtL ,

I downloaded the nighly build for Windows, and now the command-line parameters it's working as expected.

Regards.