AlbrechtL/welle.io #201: BUG: Full screen in Windows

lars18th, 2018-02-15:
Hi @AlbrechtL ,

Please, take into account that in Windows, when setting "full screen" mode, the mouse pointer is hidden. Then, without a key shortcut for restoring it, it's impossible to return. If I close the window with < ALT > + < F4 > when I restart Welle.io it allways goes full screen.

I hope you can fix this error.
Thank you!

AlbrechtL, 2018-02-16:
Which Windows version do you use?

lars18th, 2018-02-16:
> Which Windows version do you use?

Windows 7... of course!

AlbrechtL, 2018-02-17:
I can't reproduce this issue on Windows 10. Unfortunately Windows 7 is not supported.

closed 2018-02-17 by AlbrechtL

lars18th, 2018-02-17:
> Unfortunately Windows 7 is not supported.

I'm very disappointed about this!
And I don't know why.