AlbrechtL/welle.io #202: Suggestion: Cut single peaks in Spectrum

lars18th, 2018-02-15:
Hi @AlbrechtL ,

In some frequencies I see a large vertical continous line in the Spectrum. I feel it's some noise. So, I suggest to put some "top limiter" to cut the diagram when ane over peak appears.

See the attached capture:
[image]

I hope you agree to cut these peaks.
Thank you!

AlbrechtL, 2018-02-16:
You can analyze the peak with [QIRX](http://softsyst.com/qirx) which has this feature.

closed 2018-02-16 by AlbrechtL

lars18th, 2018-02-16:
> You can analyze the peak with QIRX which has this feature.

Yes, I know. But this "noise" it's permanent. So it's logical to cut-off the Spectrum.