AlbrechtL/welle.io #205: [Windows] 64bit version?

lars18th, 2018-02-16:
Hi,

I don't know why the Windows release has only a 32bit version. 64bit code in Windows it's more fast... because the large number of registers (and not because the double length of the int's). And as the Welle.io is performing a lot of DSP processing, I think it would improve efficiency by running 64-bit code.

Futhermore, a 64bit Linux version exists. So, the source it's preprared for it. And using Qt will be easy to change the target.

So, I'm wrong or a 64bit Windows version can be targeted?

AlbrechtL, 2018-02-16:
At the moment we use the MinGW build environment that comes with QT. This is only available for 32-bit.

BTW: If you compare the performance of the 64-bit Linux version and the 32-bit Windows version you will not see big performance differences.

closed 2018-02-16 by AlbrechtL

lars18th, 2018-02-16:
> At the moment we use the MinGW build environment that comes with QT. This is only available for 32-bit.

Ok. Now I know the reason. So will try (sometime) to compile it using the free MSVC.