AlbrechtL/welle.io #207: [UI] Suggestion: Programs list panel more resizable

lars18th, 2018-02-16:
Hi,

I see that the Pogram list panel is resizable **only** up to one level (around 80pixel). I found this a lot... the names never get wrapper, but when the Advanced panel is opened in several small screens the three panels are difficult to manage.

I suggest a simple change in the UI configuration for enabling the option to a more small Program list panel limit.
You agree?

AlbrechtL, 2018-02-16:
Please give use an example why 80px is to much. Which display do you use?

lars18th, 2018-02-16:
> Please give use an example why 80px is to much. Which display do you use?

- Android phone in portrait mode.
- Windows 10 tablet.
- Windows 7 in window mode ocuping only part of the monitor.

So, if this request it's only one line in the code, why not change it?
https://github.com/AlbrechtL/welle.io/blob/master/src/gui/QML/main.qml#L384
I suggest to use `Layout.minimumWidth: Units.dp(150)` or if you prefer replace with `Layout.preferredWidth: Units.dp(350)`.

_The problem is that the user can't adjust the size as s/he likes!_ 
I hope you hear users. :smile:

AlbrechtL, 2018-02-17:
I don't really understand your issue. You can make welle.io really small.

[image]

closed 2018-02-17 by AlbrechtL

lars18th, 2018-02-17:
> I don't really understand your issue. You can make welle.io really small.

I can't do this in my Windows 7 computer. Also I can't do it in my Android phone. Too in my Windows 10 Tablet. Which version are you using? Ahh!!! I see: you're using Linux Ubuntu.

In any case, any problem to change the code to `Layout.preferredWidth` instead of `Layout.minimumWidth` ? Why **force** users?