AlbrechtL/welle.io #210: Suggestion: More Technical Data in the Expert Mode

lars18th, 2018-02-18:
Hi @AlbrechtL ,

First of all: Thank you for the last release 1.0-rc3. :smile: 

One suggestion to improve the IU: Can you please add some more technical data in the Expert Mode view?

- Protection Level
- Subchannel ID
- Start Address of CU
- Used CUs
- etc

I feel all of this information is known in the code. So, it's only a requirement to print this info in the Expert Mode view. Perhaps in a "two colums mode" where you at time print SNR, Frame/RS/AAC errors, etc.

You agree?
_(I see this info in the "Technical Data" of the qt-dab tool... but your player is more robust and user friendly.)_
Regards!

andimik, 2018-02-18:
Good evening Lars,

very easy: You find that in qt-dab, just press the button `save ensemble info`

[5c.txt](https://github.com/AlbrechtL/welle.io/files/1735064/5c.txt)

lars18th, 2018-02-18:
Hi @andimik ,

Great to hear it! I'll try! :smile: 

However, I still prefer that Welle.io will "print" also this data. Why? Because this tool it's **portable**, so I can just use it with an Android device, a Windows PC or Tablet, a Mac notebook, etc. So, if I can get all the info from this great tool, then I don't need other when scanning in different regions.

In any case, @AlbrechtL has all of this info in the code: https://github.com/AlbrechtL/welle.io/blob/948fd640c6445f9e5055cbe99fc9b681f04a3a43/src/libs/faad2/libfaad/structs.h#L146
So it's only a few "prints" in the UI to show them in the current _Expert View_.
Or I'm missing something?

AlbrechtL, 2018-02-18:
Why you need this information?

lars18th, 2018-02-18:
> Why you need this information?

Why not?
And... why we need SNR, Frame errors, etc. in the Expert View? :smile: 
Not for any special reason but improve my reception in different locations.

In any case: Your DAB player is one of the most _robust_ and _portable_. So I recommend to use it in any environment. So, I like to see all the technical data in the export zone (so for this reason is this zone, true?)

andimik, 2018-02-18:
Please visit http://www.fmlist.org/sendertabelle/dab-ww.php

Here you will find all necessary information. 

If something is wrong or missing, please inform fmlist.org editors and share your observations.

lars18th, 2018-02-19:
Hi @andimik ,

> Here you will find all necessary information.

And all this info it's right (I feel). Thank you!

However, I don't like to "get" the data; I like to "see" the _decoded_ data. For example, if I try to decode some _user_ DAB bitstream (at time, modulated to IQ samples; as the ETI-NI input is not supported).

Please, think on this: The **data** exists in the App, and the info it's **relevant**. Futhermore, an **Expert** zone exists. So, what it's the reason for no show this info? Is that sensitive information?

I hope @AlbrechtL agree with printing this info. :smile:

lars18th, 2018-02-19:
Hi @mpbraendli ,

Great idea! :+1: 

However, before show complex data (like the occupued CUs), will be sufficient to print the _numerical_ or _string_ data, like the Protection Level, Subchannel ID, Start Address of CU, _number of_ Used CUs, etc.

@AlbrechtL , you agree?

andimik, 2018-02-19:
For such a reason please use Andi Gsinn's player, he shows every CU in one little square 
http://www.ukwtv.de/cms/downloads-aside/281-dab-player-von-andreas-gsinn.html
So there is already a great software for it. No need to implement it at welle.io

mpbraendli, 2018-02-19:
@andimik I cannot use Andi Gsinn's player, because I don't have a Windows machine.
Also, I disagree that you take the existence of a feature inside another player as an argument to not implement the feature in welle.io. Taking this position further, you could also say "why develop welle.io? please use Andi Gsinn's player instead".
That's not helping us answer the question "what do we want to see in the expert view"? I would love to see lots of expert information (some of which are already on the TODO list, like TII and CIR)

andimik, 2018-02-19:
Lars uses Windows 7 ;-)

AlbrechtL, 2018-03-03:
I added it to the [Open Tasks](https://github.com/AlbrechtL/welle.io/wiki/Open-Tasks).

closed 2018-03-03 by AlbrechtL